dora-rs/dora #616: Receiving multiple Input from Web portal.

XiaoKuge, 2024-08-08:
**Is your feature request related to a problem? Please describe.**
We are developing a Web frontend that drives dora-flows at the backend. It seems that we can only setup a dynamic node at the begining of the dora flow to get input. The middle nodes won't be able to take the additional input.

**Describe the solution you'd like**
Provide the middle-nodes capability to receive input from frontend.

**Describe alternatives you've considered**
N/A

**Additional context**

Thank you! 

from MAE team

haixuanTao, 2024-08-08:
I'm not sure, I understand. Could you provide a simple example ?

chengzi0103, 2024-08-09:
> I think that you might have been blocked by: #606 not being merged.
> 
> Now that it's merged it should be easier to have receive-only dynamic node

How should I use this configuration? Could you provide an example of how to configure synchronized_start in YAML or Python?

haixuanTao, 2024-08-09:
Do you need that all nodes are ready to start or do you want them to start independently from your web front end?

chengzi0103, 2024-08-09:
> Do you need that all nodes are ready to start or do you want them to start independently from your web front end?

I want them to start independently from the web front end.

haixuanTao, 2024-08-09:
Yes, so you don't need to configure anything the new merged PR will allow you to do just that.

chengzi0103, 2024-08-09:
> Yes, so you don't need to configure anything the new merged PR will allow you to do just that.

However, have you tried including multiple dynamic nodes in a dataflow? When I started one of the nodes, there was no response at all.

Can you take a look at this issue?

https://github.com/dora-rs/dora/issues/607

haixuanTao, 2024-08-09:
Yes, with the latest version that I have merged. You can include multiple dynamic nodes. You can even include them in the same python scripts.

chengzi0103, 2024-08-09:
> Yes, with the latest version that I have just merged. You can include multiple dynamic nodes. You can even include them in the same python script.

How should I get your updates? Should I reinstall Dora-RS? If so, which version should I install?

haixuanTao, 2024-08-09:
Yes so, to update to latest main:

```bash
sudo rm $(which dora)
pip uninstall dora-rs

## dora binary
git clone https://github.com/dora-rs/dora.git
cd dora
cargo build --release -p dora-cli
PATH=$PATH:$(pwd)/target/release

## Python API
maturin develop -m apis/python/node/Cargo.toml

dora --help
```

chengzi0103, 2024-08-09:
> Yes so, to update to latest main:
> 
> ```shell
> sudo rm $(which dora)
> pip uninstall dora-rs
> 
> ## dora binary
> git clone https://github.com/dora-rs/dora.git
> cd dora
> cargo build --release -p dora-cli
> PATH=$PATH:$(pwd)/target/release
> 
> ## Python API
> maturin develop -m apis/python/node/Cargo.toml
> 
> dora --help
> ```

Alright, thank you.

closed 2025-03-18 by Hennzau