dora-rs/dora #617: Send multiple outputs to the frontend

XiaoKuge, 2024-08-08:
**Is your feature request related to a problem? Please describe.**
We have a frontend. While a dora-flow is running, we hope to be able to show outputs data from the middle nodes. 

**Describe the solution you'd like**
Middle nodes can send data easily to the external environment. 

**Describe alternatives you've considered**
N/A

**Additional context**
Thank you and best regards,

from MAE team

haixuanTao, 2024-08-08:
Could you clarify the meaning of showing output from the middle node ? 

What do you use to show output ?

XiaoKuge, 2024-08-09:
[image]
Assuming the psudo YAML script above, there is a "middle-node", and somehow we want it output something to Web browser (frontend).

Hope it helps. 

Thank you.

haixuanTao, 2024-08-09:
So, currently we do not provide any web browser based API ( javascript, ...) 

However we could help provide a REST API for retrieving logs from opentelemetry, if that works for you?

haixuanTao, 2024-08-31:
Closing this as I think the problem should be mostly solved with dynamic node.

closed 2024-08-31 by haixuanTao