dora-rs/dora #618: Questions regarding the dynamic-node.

chengzi0103, 2024-08-09:
**Is your feature request related to a problem? Please describe.**

Regarding the dynamic node, we currently have the following issues:

1. We want the `dynamic-node` to retrieve the output of a specific node, then use `python dynamic.py` to display the received data in the `terminal`.
2. When we have multiple `dataflow` processes and the `dynamic` node names are the same across these processes, why can't we start the same `dynamic` node to send information to multiple `dataflow` processes?

haixuanTao, 2024-08-09:
2. A node cannot be used for multiple dataflows. You should either rewrite your dataflow to be a single dataflow or try to use multiple dynamic nodes from a single python to make two dataflow collaborate. 

But I would recommend focusing on a single dataflow for now.

1. So is there an issue with printing into the terminal?

chengzi0103, 2024-08-09:
> 2. A node cannot be used for multiple dataflows. You should either rewrite your dataflow to be a single dataflow or try to use multiple dynamic nodes from a single python to make two dataflow [...]

Let's discuss the first issue.

When I run the code below, I input a task in the terminal, but the task is not sent to the dataflow, and the dynamic node does not terminate. I'm not sure if there's an issue with my code.

```python
import json
from dora import Node
import pyarrow as pa

node = Node("reasoner_task_input")

event = node.next()
task_data = input('Please enter your task: ')
node.send_output('reasoner_task', pa.array([json.dumps(task_data)]), event["metadata"])

while True:
    event = node.next()
    if event["type"] == "INPUT":
        if event["id"] == "reasoner_log":
            print('-----------  ：', event["value"])
```

haixuanTao, 2024-08-09:
Could you share `out/dora-daemon.txt` and `out/dora-coordinator.txt`?

chengzi0103, 2024-08-09:
> Could you share `out/dora-daemon.txt` and `out/dora-coordinator.txt`?

None of the `out/*.txt` files in this process contain any results, and the dataflow itself hasn't been executed.

haixuanTao, 2024-08-09:
ok could you run:

```bash
dora destroy 
dora up
```

chengzi0103, 2024-08-09:
> ```shell
> dora destroy 
> ```

1. The result is still that the dora-dataflow wasn't triggered.
2. I'm not sure why the `dynamic` node is receiving the `reasoner_log` message.

Below is the content of my dataflow YAML file:

```yaml
nodes:
  - id: reasoner_task_input
    path: dynamic
    inputs:
      direction: dora/timer/secs/1
      reasoner_log: reasoner_output/reasoner_output
    outputs:
      - reasoner_task
  - id: reasoner_agent
    operator:
      python: ./scripts/reasoner_agent.py
      inputs:
        reasoner_task: reasoner_task_input/reasoner_task
      outputs:
        - reasoner_result
  - id: reasoner_output
    operator:
      python: ./scripts/reasoner_output.py
      inputs:
        reasoner_result: reasoner_agent/reasoner_result
      outputs:
        - reasoner_output
```

---

chengzi0103, 2024-08-09:
> 1. The result is still that the dora-dataflow wasn't triggered.
> 2. I'm not sure why the `dynamic` node is receiving the `reasoner_log` message.
>
> Below is the content of my dataflow YAML file: [...]
翻译：

This is the received content of `task_input`:

```
python3 reasoner_task_input.py
Please enter your task:  Hello
-----------  ： [
  ""{\"task\": \"Hello\", \"result\": \"Select only one question\"}""
]
```

haixuanTao, 2024-08-09:
So is this the result you were expecting?

I have opened: https://github.com/dora-rs/dora/pull/620 to make it simpler for people to send data from the terminal and print the data that is received.

chengzi0103, 2024-08-12:
> So is this the result you were expecting?
> 
> I have opened: #620 to make it simpler for people to send data from the terminal and print the data that is received.

This is the result I was looking for, thank you!

However, I couldn't find the examples/terminal-dataflow example. Could you also tell me how to install the terminal-print command?

haixuanTao, 2024-08-12:
Make sure to use the branch `terminal-example`

```bash
git checkout terminal-example
```

So when you use:
```bash
dora build dataflow.yml
```
 It should install `terminal-print`

haixuanTao, 2024-08-31:
This has been solved by the addition of the echo example, the terminal-input and terminal-print in the node-hub

closed 2024-08-31 by haixuanTao