dora-rs/dora #621: Metadata problem in python-dataflow

JojoDevel, 2024-08-09:
**Describe the bug**
The `python-dataflow` examples fails because the `metadata` field does not contain any useful information.

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. Go to the dataflow example `cd examples/python-dataflow`
3. Install dependencies `dora build dataflow.yml`
4. Start the dataflow: `dora start dataflow.yml`

**Expected behavior**
Launch of python-dataflow example.

**Screenshots or Video**

Launching the dataflow fails because the message metadata is not available:
[image]

If we log the `metadata` field of the message, we see that it is almost empty: ```{'open_telemetry_context': ''}```

**Environments (please complete the following information):**
 - System info: [use `uname --all` on Linux]
   `Linux [PC-name] 6.5.0-44-generic #44~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Jun 18 14:36:16 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

 - Dora version: [use `dora --version` and `pip show dora-rs`]
  ```
  dora-cli 0.3.5
  ```
  ```
  Name: dora-rs
  Version: 0.3.5
  Summary: `dora` goal is to be a low latency, composable, and distributed data flow.
  Home-page: 
  Author: 
  Author-email: 
  License: Apache-2.0
  Location: /home/johannes/miniconda3/envs/dora-env/lib/python3.10/site-packages
  Requires: pyarrow
  Required-by: opencv-plot, opencv-video-capture, ultralytics-yolo
  ```

Hennzau, 2024-08-10:
Hi! I think that's because you are using Dora release 0.3.5 version, and the examples from the repository are using the latest main branch version.

What you should do is:
- uninstall dora-cli `cargo uninstall dora-cli`
- go inside your clone of Dora repository, and `cargo install .`

Those steps will install the latest main version with the Custom metadata features

JojoDevel, 2024-08-11:
Thanks a lot :+1:  It works for me by using the latest main branch version. Didn't know that the `metadata` is among the latest features.

closed 2024-08-11 by JojoDevel

Hennzau, 2024-08-11:
> Thanks a lot :+1:  It works for me by using the latest main branch version. Didn't know that the `metadata` is among the latest features.

Metadata are not new features, Custom metadata with user defined fields are! You're welcome