dora-rs/dora #642: Node panic on `timeout while waiting for event stream thread`

haixuanTao, 2024-08-30:
Not exactly sure what is happening, but it seems that I have a bug where the node completely freeze and then report:

```bash
[2m2024-08-30T16:15:37.258067Z[0m [33m WARN[0m [2mdora_node_api::event_stream::thread[0m[2m:[0m timeout while waiting for event stream thread
```

Then the node finishes. 

I wonder if there is like a break condition on: https://github.com/dora-rs/dora/blob/9be5b765b94a8212011f3cb6e5b4e74f2f487269/apis/rust/node/src/event_stream/thread.rs#L70-L72

[Screencast from 2024-08-30 18-29-53.webm](https://github.com/user-attachments/assets/460f2a63-6cf9-41c3-9e70-1597934cf3ac)

haixuanTao, 2024-08-30:
This only happens with python by the way. Does not happen with rust `dora-rerun`

haixuanTao, 2024-08-31:
FYI, this seems to happen when there is more than 1 listener of the camera output stream.

Rustix69, 2025-03-07:
@haixuanTao Can I work on this Issue ? Plz assign me these issues I can get it done by tommorow

Hennzau, 2025-03-18:
> [@haixuanTao](https://github.com/haixuanTao) Can I work on this Issue ? Plz assign me these issues I can get it done by tommorow

I assigned you this issue. Tell me if you finally don't want to work on it

phil-opp, 2026-01-31:
Is this issue still reproducible?