dora-rs/dora #648: coordinator crashes on bad logs access

Hennzau, 2024-09-09:
**Describe the bug**

When I try to view logs using `dora logs`, the coordinator crashes if I input incorrect information (e.g., an invalid UUID or name). The same happens when there is an error, such as a wrong node ID or when multiple dataflows have the same name.

**Environment (please complete the following information):**
- System: macOS
- Dora version: tag v0.3.6

Hennzau, 2024-09-10:
The same issue occurs with some other commands (`dora stop [uuid]` with an invalid UUID). This is due to the [coordinator control event loop](https://github.com/dora-rs/dora/blob/69f0526fe2c82945ffe02821ded3303347407f13/binaries/coordinator/src/lib.rs#L304).

```rust
ControlRequest::Logs { uuid, name, node } => {
    let dataflow_uuid = if let Some(uuid) = uuid {
        uuid
    } else if let Some(name) = name {
        resolve_name(name, &running_dataflows, &archived_dataflows)?
    } else {
        bail!("No UUID provided")
    };

    let reply = retrieve_logs(
        &running_dataflows,
        &archived_dataflows,
        dataflow_uuid,
        node.into(),
        &mut daemon_connections,
        clock.new_timestamp(),
    )
    .await
    .map(ControlRequestReply::Logs);
    let _ = reply_sender.send(reply);
}
```

I don't think we should stop the coordinator if we can't resolve the dataflow UUID. I'll wrap it so it just sends an **error reply** instead.

Now it looks like that:
```rust
ControlRequest::Logs { uuid, name, node } => {
let dataflow_uuid = if let Some(uuid) = uuid {
    Ok(uuid)
} else if let Some(name) = name {
    resolve_name(name, &running_dataflows, &archived_dataflows)
} else {
    Err(eyre!("No uuid"))
};

match dataflow_uuid {
    Ok(uuid) => {
        let reply = retrieve_logs(
            &running_dataflows,
            &archived_dataflows,
            uuid,
            node.into(),
            &mut daemon_connections,
            clock.new_timestamp(),
        )
        .await
        .map(ControlRequestReply::Logs);
        let _ = reply_sender.send(reply);
    }
    Err(err) => {
        let _ = reply_sender.send(Err(err));
    }
}
```
And in case of bad logs name/uuid it looks like that: 
```bash
[ERROR]
unexpected reply to daemon logs: Error("no dataflow with name `test`")
```

haixuanTao, 2024-09-10:
Yes. I agree that this should not happened. And this looks good!

Could you open a PR?

Thanks in advance :)

PR #650 (2024-09-11): Coordinator stopped on bad control command
Fix #648 and more
closed 2024-09-11 by Hennzau