dora-rs/dora #651: High CPU Usage in the Dora Environment When Running SLAM with PCL Library

ZcanMelo, 2024-09-10:
I am running the same SLAM algorithm code on the NVIDIA Orin platform using both Dora and ROS environments. 
[image]
[image]
The first image shows the CPU usage in the ROS environment, while the second image shows the CPU usage in the DORA environment.

Despite identical sensors and controllers, the CPU usage is significantly higher in the Dora environment. The high CPU consumption has been traced to the align function from the PCL library. 
[image]

I am seeking to understand the reason behind this difference in CPU usage and is considering whether it could be due to compatibility issues between the Dora environment and the PCL library.

Hennzau, 2024-09-10:
Hi! thanks for the report, can you tell us what is your dora version and how did you installed this version?

ZcanMelo, 2024-09-10:
Hi! My Dora version is 0.3.2. I installed it via the Releases page, downloading both the zip and whl files.

Hennzau, 2024-09-10:
Okay, so 0.3.2 is old, can you give a try with 0.3.6?

ZcanMelo, 2024-09-11:
I received it, thank you for your response, I will try it.

ZcanMelo, 2024-09-12:
Hello, thank you for your reply. I have already tried commenting out the print statements, but the effect was not significant.

haixuanTao, 2024-09-19:
Are you sure, you're running the same frequency?

Hennzau, 2025-03-18:
@ZcanMelo did you have time to try again with a new dora version?

ZcanMelo, 2025-03-18:
@Hennzau Thank you for your suggestion. I'm currently using the Dora version 0.3.8, and at present, the CPU occupancy rate of using Dora is similar to that of ROS1.

closed 2025-03-18 by Hennzau