dora-rs/dora #694: Dataflow could not be validated for Getting Started

jmskov, 2024-10-23:
**Describe the bug**
I have installed dora according to the instructions and Getting Started section in the README. I can build the example python-dataflow fine. I get the error `[ERROR] Dataflow could not be validated.` when running the example.

**To Reproduce**
Steps to reproduce the behavior:
0. Install `dora-cli` using cargo, activate venv in examples/python-dataflow
1. Run `dora up`
2. Run `dora build dataflow.yaml`
3. Run `dora start dataflow.yaml`

**Expected behavior**
The webcam example to run.

**Environments (please complete the following information):**
 - System info: MacOS 15.1, M2Pro. 
 - Dora version:  dora-cli 0.3.6, 
Name: dora-rs
Version: 0.3.6
Summary: `dora` goal is to be a low latency, composable, and distributed data flow.
Home-page: 
Author: 
Author-email: 
License: Apache-2.0
Location: /Users/john/Projects/rust/dora/dora/examples/python-dataflow/meow/lib/python3.12/site-packages
Requires: pyarrow
Required-by: dora-yolo, opencv-plot, opencv-video-capture

**Additional context**
Using Python 3.12.7 and pip via homebrew. I can confirm `import cv2; cap = cv2.VideoCapture(0)` can capture my webcam from the interpreter in the venv. I suspect it is security related, but not sure where.

haixuanTao, 2024-10-24:
Ok, interesting, I wonder if it's because, the environment path management for python in macOS is different.

Going to double check this and probably change the example. 

Thanks for reporting this issue.

haixuanTao, 2024-10-24:
Do you mind trying out the `v0.3.7rc0`  version?

You can install the new version with:

```bash
curl --proto '=https' --tlsv1.2 -sSf https://raw.githubusercontent.com/dora-rs/dora/main/install.sh | bash -s -- --tag v0.3.7rc0

source ~/.bashrc
```

Or download it manually at: https://github.com/dora-rs/dora/releases/tag/v0.3.7rc0

jmskov, 2024-10-24:
Same result. Also not sure if it is an oversight, but the crate version was not changed in the release candidate. I will try some other ideas and update if I make progress.

haixuanTao, 2024-10-24:
Ok sorry. 

Yeah, I wanted to do a quick release candidate without having to change all package version conflict.  Maybe `v0.3.6-post` naming would have made more sense.

Those path dependency are very hard to predict.

I want to add a test suite within dora to simplify testing. 

Basically, you can try to do:
```
pip install opencv-video-capture opencv-plot
# And try individually:
opencv-video-capture

# It it returns a path error it means that the installation didn't work
# If it returns an error with `DORA_NODE_CONFIG` it means that it should work, but the path couldn't be validated.
```

jmskov, 2024-10-24:
No worries, just wanted to check 😄 I believe my error is the latter, here is my trace when running `opencv-video-capture`:

```
File "/Users/john/Projects/rust/dora/dora/examples/python-dataflow/bin/opencv-video-capture", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/john/Projects/rust/dora/dora/examples/python-dataflow/lib/python3.12/site-packages/opencv_video_capture/main.py", line 74, in main
    node = Node(args.name)
           ^^^^^^^^^^^^^^^
RuntimeError: Could not setup node from node id. Make sure to have a running dataflow with this dynamic node

Caused by:
    failed to get node config from daemon: no node with ID `{node_id}`

Location:
    apis/rust/node/src/node/mod.rs:106:17

haixuanTao, 2024-10-24:
Ok, so I think the path resolver from the cli did not work, which is somehow surprising. As it should work ... in theory.

Would you mind:

```bash
dora destroy 
dora up
dora start dataflow.yaml
```

jmskov, 2024-10-24:
That fixed it. I tried a few permuations to replicate the issue (e.g. dora up in new terminal, outside of the env), but I have not been able to.

haixuanTao, 2024-10-24:
Yeah, it's actually a fairly common of issue that I hope to resolve with #691 

Thanks for taking the time.

haixuanTao, 2025-01-02:
So FYI this error was linked to python3 linking to the wrong python that was fixed in https://github.com/dora-rs/dora/pull/731 . Sorry for this.

closed 2025-01-02 by haixuanTao