dora-rs/dora #708: Dora-Message is not independant from `dora-core` crate which create conflict when trying to publish a new dora version

haixuanTao, 2024-11-11:
**Describe the bug**
Currently `dora-message` depends on a fixed version of `dora-core` which means that publishing a new version of `dora` with requirement on newly added `dora-core` version will conflict with the pinned `dora-core` version of `dora-message.

Related to: https://github.com/dora-rs/dora/pull/614 

## TODO:

We need to make dora-message independant from `dora-core` if we want to publish a new version of dora.

**To Reproduce**
Check: https://github.com/dora-rs/dora/actions/runs/11664027456/job/32473707635

phil-opp, 2024-11-11:
Interesting! I think in this specific case the issue only occurs because you're trying to publish a pre-release, which is not valid for the `0.3.6` version requirement. My guess is that it would work for the actual `0.3.7` release (i.e. not a pre-release). 

However, we should be able to do pre-releases and the tight coupling of `dora-core` and `dora-message` can probably lead to other problems too. So I agree that we should remove the `dora-core` dependency from `dora-message`.

haixuanTao, 2024-11-12:
I mean from the error message:
```

Caused by:
  failed to select a version for `dora-core`.
      ... required by package `dora-message v0.4.0`
      ... which satisfies dependency `dora-message = "^0.4.0"` of package `dora-node-api v0.3.7-rc1 (/home/runner/work/dora/dora/target/package/dora-node-api-0.3.7-rc1)`
  versions that meet the requirements `^0.3.6` are: 0.3.6

  all possible versions conflict with previously selected packages.

    previously selected package `dora-core v0.3.7-rc1`
      ... which satisfies dependency `dora-core = "^0.3.7-rc1"` of package `dora-node-api v0.3.7-rc1 (/home/runner/work/dora/dora/target/package/dora-node-api-0.3.7-rc1)`

  failed to select a version for `dora-core` which could resolve this conflict
```

`dora-message=0.4.0` was released with the 0.3.6 release pinning its `dora-core` version to `0.3.6`. The only fix would be to release a `0.4.1` version pinning `dora-core` to `0.3.7`.

I don't think it is a pre-release issue.

PR #711 (2024-11-14): Move more types from `dora-core` to `dora-message` to avoid dependency
Make `dora-message` a dependency of `dora-core`, instead of the other way around. This way, we can continue to freely bump the version of `dora-core` with the other workspace crates, without introducing errors such as #708.

Fixes #708