dora-rs/dora #727: Orange Pi Ai Pro 20T memory allocation failed

LyonRust, 2024-12-03:
**Orange Pi Ai Pro 20T memory allocation failed**
When connecting remotely using dora, start the robot acquisition data node, and when the rerun interface pops up, orangepi's dora node holds an error

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start coordinator: `dora coordinator` 
2. Dora start daemon: `dora daemon -c 192.168.1.5`
3. ssh login in orangepi and staer daemon `dora daemon -c 192.168.1.5 --machins-id ec7`
4. Start a new dataflow: `dora start qwen2_rebot_recorder.yml`

**Expected behavior**
on the prangepi error
```
memory allocation of 7885080994159069819 bytes failed
Aborted (core dumped)
```

**Screenshots or Video**
video： https://github.com/user-attachments/assets/5fd79c35-0af6-48cf-a4bc-9d93e003e535

**Environments (please complete the following information):**
 - System info: [use `uname --all` on Linux]
 - `Linux orangepiaipro-20t 5.10.0+ #1 SMP Tue Jun 4 14:51:23 CST 2024 aarch64 aarch64 aarch64 GNU/Linux`
 - Dora version: [use `dora --version` and `pip show dora-rs`]
 - dora-cli 0.3.7

**Additional context**
qwen2_robot_recorder.yml
```yaml
nodes:
  - id: arm # Run on car
    #build: wget -nc https://github.com/dora-rs/gosim-2024/releases/download/v0.0.1/robot # cargo install --git https://github.com/lyon-dora/chongyou-car
    path: dora-gen72
    _unstable_deploy:
      machine: ec7
    inputs:
      movec: arm-interpolation/movec
      movej: arm-interpolation/movej
      claw: arm-interpolation/claw
      save: arm-interpolation/save
      go_to: arm-interpolation/go_to
    env:
      ROBOT_IP: 192.168.1.18

  - id: mickrobot_chassis # Run on car
    #build: wget -nc https://github.com/dora-rs/gosim-2024/releases/download/v0.0.1/robot # cargo install --git https://github.com/lyon-dora/chongyou-car
    path: robot
    _unstable_deploy:
      machine: ec7
    inputs:
      text: dora-qwenvl-recorder/text
    env:
      SERIAL_PORT: /dev/ttyUSB0

  - id: keyboard-listener # Run on car
    #build: pip install dora-keyboard
    path: dora-keyboard
    inputs:
      tick: dora/timer/millis/10
    outputs:
      - char
    env:
      DISPLAY: :0

  - id: key-interpolation
    path: key_interpolation.py
    inputs:
      keyboard: keyboard-listener/char
    outputs:
      - text

  - id: arm-interpolation
    path: arm_interpolation.py
    inputs:
      keyboard: dora-qwenvl-recorder/text
    outputs:
      - movec
      - movej
      - claw
      - save
      - go_to

  - id: camera1 # Run on car
    #build: pip install opencv-video-capture
    path: opencv-video-capture
    _unstable_deploy:
      machine: ec7
    inputs:
      tick: dora/timer/millis/50
    outputs:
      - image
    env:
      CAPTURE_PATH: 0
      IMAGE_WIDTH: 640
      IMAGE_HEIGHT: 480
      ENCODING: jpeg
      FLIP: BOTH

  - id: camera2 # Run on car
    #build: pip install opencv-video-capture
    path: opencv-video-capture
    _unstable_deploy:
      machine: ec7
    inputs:
      tick: dora/timer/millis/50
    outputs:
      - image
    env:
      CAPTURE_PATH: 2
      IMAGE_WIDTH: 640
      IMAGE_HEIGHT: 480
      ENCODING: jpeg
      FLIP: BOTH

  - id: plot
    #build: wget -nc https://github.com/dora-rs/gosim-2024/releases/download/v0.0.1/rerun
    #build: wget -nc https://github.com/dora-rs/gosim-2024/releases/download/v0.0.1/dora-rerun
    path: dora-rerun
    inputs:
      image_1:
        source: camera1/image
        queue_size: 1
      image_2:
        source: camera2/image
        queue_size: 1
      textlog_vlm: dora-qwenvl-recorder/text
    env:
      IMAGE_WIDTH: 640
      IMAGE_HEIGHT: 480
      RERUN_MEMORY_LIMIT: 25%
      README: |
        To control the robot, use the following keys:
        - w: move forward
        - s: move backward
        - a: move left
        - d: move right
        - q: stop
        - r: open gripper
        - t: close gripper
        - h: move arm left
        - k: move arm right
        - y: move arm down
        - i: move arm up
        - u: move arm forward
        - j: move arm backward
        - v: go home
        - b: save home

  - id: dora-qwenvl-recorder
    path: llama-factory-recorder
    inputs:
      image_1:
        source: camera1/image
        queue_size: 1
      image_2:
        source: camera2/image
        queue_size: 1
      ground_truth: key-interpolation/text
    outputs:
      - text
    env:
      DEFAULT_QUESTION: Respond with arm left, arm right, arm up, arm down, open, close, or go home to Followed by a man in white
      USE_MODELSCOPE_HUB: True
      LLAMA_FACTORY_ROOT_PATH: /home/demo/LLaMA-Factory
```

haixuanTao, 2024-12-03:
7885080994159069819 bytes correspond to 7344.13 GB, . I feel like there's probably an issue, with the data you're trying to send or receive.

LyonRust, 2024-12-04:
Strange question came up and today everything is normal again. No changes to any of the code.

closed 2024-12-04 by LyonRust