dora-rs/dora #737: Node frequency impacting other nodes?

pierre-rouanet, 2024-12-30:
Hi! 
I've started playing a bit with dora to better understand its possibilities. It's really interesting and promising!

Based on the talker/listener examples, I've made one talker really high-frequency (eg. 500Hz) and kept the other at 1Hz. 
The high-frequency node naturally uses quite some CPU. But the other talker with the low-frequency also quite significantly increases its CPU consumption (~ 20x). 

Are all nodes receiving the tick updates of the highest frequency? Is there something I should do differently?
I've also tried mixing Python and Rust nodes as ideally, I would like to have some high frequency nodes running in Rust (typically for low-level motor control) and higher level nodes in Python.

haixuanTao, 2025-01-02:
Hello Pierre!

Thanks for trying it out!

We do indeed have a lot of CPU-inefficiency while receiving data that we haven't optimized on in order to have the least latency on reception.

We will work on this in order to achieve high-frequency 500Hz, but at is currently implemented. High frequency is currently not managed by dora unfortunately.

Your vision on rust-python mixing is exactly what we want to do, and is definitely in our roadmap.

pierre-rouanet, 2025-01-02:
Great, thank you Tao for your answer!
I'll probably investigate a bit more later as we are considering alternatives option than using ROS2 as the core framework for our robots.
Let me know if we can help!