dora-rs/dora #747: Regarding the delivery of events in Dora.

chengzi0103, 2025-01-07:
In the add-numbers node, I use a while True loop to continuously retrieve the value of num2. After successfully obtaining the value of num2, I attempt to use the same method to retrieve the value of num1, but I find that num1 is no longer available and has become None. Could you provide a detailed explanation of the event delivery mechanism and its underlying logic?

**Environments**

- System info: mac os 
- Dora version: 0.3.8

**Additional context**

The dataflow configuration is as follows:

```yaml
nodes:
  - id: terminal-input
    build: pip install -e ../../node-hub/terminal-input
    path: dynamic
    outputs:
      - data
    inputs:
      add_numbers_result: add-numbers/add_numbers_result
      multiply_numbers_result: multiply-numbers/multiply_numbers_result
  - id: multiply-numbers
    build: pip install -e ../../agent-hub/multiply-numbers
    path: multiply-numbers
    outputs:
      - multiply_numbers_result
    inputs:
      num1: terminal-input/data
    env:
      IS_DATAFLOW_END: false
  - id: add-numbers
    build: pip install -e ../../agent-hub/add-numbers
    path: add-numbers
    outputs:
      - add_numbers_result
    inputs:
      num1: terminal-input/data
      num2: multiply-numbers/multiply_numbers_result
    env:
      IS_DATAFLOW_END: true
```

1. **Add Logging:**
add-numbers-nodes-logs:

   num2 :  40
   num1 :  None

ayushshrivastv, 2025-03-01:
To fix this, store num1 when it is first received (using the on_input callback) and wait for num2 to arrive before performing the addition. Avoid while True loops; rely on event triggers to manage state. Ensure nodes handle inputs asynchronously, preserving data between events for correct computation.

ayushshrivastv, 2025-03-01:
What’s happening here is that when the first number is received from the terminal input or data, it’s consumed by the add-numbers function. By the time the second number, num2, arrives, num1 is no longer available, so it returns None.

haixuanTao, 2026-01-30:
I think this might have been fixed with a queue_size