dora-rs/dora #748: Interactive and realtime visualisation with gradio

haixuanTao, 2025-01-07:
### Discussed in https://github.com/orgs/dora-rs/discussions/746

<div type='discussions-op-text'>

<sup>Originally posted by **jrabary** January  6, 2025</sup>
Hi all,

I'm trying to add an interactive and realtime visualisation node in my graph. I end up having the following events handlers in the GUI 

```
    def handle_submit(command):
        if command != "":
            instruction_array = pa.array([command])
            node.send_output("instruction", instruction_array)
        return ""

    def handle_camera_streams():
        current_im = im
        while True:
            evt = node.next(timeout=FPS)
            if evt is not None:
                if evt["type"] == "INPUT":
                    if evt["id"] == "observations":
                        obs = decode_dict(evt["value"])
                        current_im = obs["overhead_cam"]

            yield current_im
```

The first one is to handle user action on the GUI and the second on to display a video stream on the GUI. The point is, the two handlers can work separatly but once I put them together I got some errors

```
node.send_output("instruction", instruction_array)
RuntimeError: Already borrowed
```

I suspect dora doesn't like I use the same node object concurrently. 

Any ideas ?</div>

7SOMAY, 2025-03-18:
Hi @haixuanTao, I believe this issue can be handled using multi-threading in Python.
We can create separate threads for each functionality to process them simultaneously without blocking.

I guess this will work, what do you think?

haixuanTao, 2025-03-18:
I think that we currently have an issue if we try to call in parallel dora from different thread, but haven´t really investigated deeply with this.

7SOMAY, 2025-03-19:
> I think that we currently have an issue if we try to call in parallel dora from different thread, but haven´t really investigated deeply with this.

Oh I see, if you want-I can give it a try. Just let me know the reference and a kick start to grasp the exact situation to work on.

7SOMAY, 2025-03-19:
> Honestly. it's quite an intricate issue as it is deeply rooted within the mixed rust-python side of dora.

Just let me have a quick overview of that, I’ll try to get my hands on that.
At least I’ll learn something. 

But only if you say so, o/w I’ll keep working on my existing issues.

haixuanTao, 2025-03-19:
Being bluntly honest, I don't even know myself how to fix it! XD 

I'm not even sure where does this come from. I genuinely think that it can take about a week to fix

7SOMAY, 2025-03-19:
Okay got it, if you find any lead to this. Please raise detailed descriptions, till then I'll work on pending issues and try to get a little bit of understanding of this issue as well :)

haixuanTao, 2025-11-30:
I think one of the + for 1. is that it enables concurrent sending of event which is often a use case of concurrent calls rather than just concurrent read and write.

I wonder if this needs to be under a feature flag for people who might not want to add this global lock call.

haixuanTao, 2025-12-02:
Closed as of #1244