dora-rs/dora #775: some questions for camera

Jia-Baos, 2025-02-10:
When I start a node for capturing images from a camera, and then I want to change the camera's IP and port while it is running without restarting the node, how can I achieve this effectively? Or can Dora achieve this function?

haixuanTao, 2025-02-11:
There currently no way to do such things, unfortunately...

Jia-Baos, 2025-02-12:
> There currently no way to do such things, unfortunately...

Thanks, in addition, there seems to be some issues with multi-machine communication. Could you update the relevant user documentation?

haixuanTao, 2025-02-12:
So, we currently have an active branch: https://github.com/dora-rs/dora/tree/zenoh-daemon that is trying to fix those issues mostly documented here: https://github.com/orgs/dora-rs/discussions/699