dora-rs/dora #780: Coordinator does not have `--uv` option

Mivik, 2025-02-13:
**Is your feature request related to a problem? Please describe.**
While `dora run` has `--uv` option to run dataflow with local uv environment, `dora daemon` (or `dora up`) does not have this option, causing confusion since currently failure to spawn nodes would yield no error message in CLI.

**Describe the solution you'd like**
Ideally, `--uv` should not be an option (except for `dora build`). The current behavior of `--uv` is to prepend `uv run` to the spawn command. This could, however, be replaced with running `. .venv/bin/activate` beforehand. This method applies to both `dora up` and `dora run`, and IMO should be the preferred way in documentation. The old `run --uv` could be deprecated but still usable.

haixuanTao, 2025-02-13:
Hey !

Thanks for reaching out!

Indeed coordinator should be able to pass uv flag. We added uv very recently and wanted to try it out before changing everything. I think Philipp is working on a big PR at #779 and after this we can add uv into the coordinator. Is that ok? Sorry again for this. I maybe should have added a warning.

I would also want to not use the `--uv` flag but I do feel it can be confusing at times:
- If `uv` choose an environment that is different from the one within your terminal
- If `uv` is not installed for whatever reason.
- we merged this very recently and so we might not want to introduce some unknown bugs.

But, super happy if uv is a critically good feature for you.

phil-opp, 2025-02-13:
Instead of adding it at the coordinator level, it probably makes more sense to add it to the `dora start` command of the CLI, doesn't it? Because the coordinator doesn't run any of the Python code.

haixuanTao, 2025-02-13:
Yeah, indeed at the start level.

haixuanTao, 2025-02-13:
Adding to start, help us simplify a lot our CI/CD by removing all virtual environments activation logic.

closed 2025-02-23 by haixuanTao