dora-rs/dora #806: Fix Clippy Linting

haixuanTao, 2025-03-05:
To test linting, just do:

```bash
cargo clippy --all --tests --examples 
```

Most of the times, clippy would propose changes that can be easily accepted. :)

MunishMummadi, 2025-03-07:
Hey @haixuanTao can i work on this issue. 
My understanding is that you just want me to run the following commands and push the code :)
```console 
cargo clippy --all --tests --examples
```
```console
cargo clippy --all --tests --examples  --fix
```
or do you want to me to address the issues as produced by the `compiler`

haixuanTao, 2025-03-07:
Ahah, it is indeed, not very complicated 😊

But let's open a PR, and see if the CI/CD raise any unexpected warnings 😊

It's okay if the first PR do not fix all issues.

Rustix69, 2025-03-07:
@haixuanTao Can I work on  the issue ?

haixuanTao, 2025-03-07:
In order to move forward with this issue.

We need to fix depreciation coming from the change of pyo3 0.23:

> PyO3 0.23 completes the removal of the "GIL Refs" API in favour of the new "Bound" API introduced in PyO3 0.21.

> With the removal of the old API, many "Bound" API functions which had been introduced with _bound suffixes no longer need the suffixes as these names have been freed up. For example, PyTuple::new_bound is now just PyTuple::new (the existing name remains but is deprecated).

To do this, could anyone, remove `_bound` suffix in all concerned rust code?

```bash
# The vim command would look like this
s/_bound//g

# Similarly replace:
s/to_object/into_pyobject/g

# Similarly replace:
s/to_py/into_pyobject/g
```

Thanks in advance. 

To go even further we could modify: https://github.com/huggingface/pyo3-special-method-derive with the same changes. :)

MunishMummadi, 2025-03-07:
@haixuanTao Is there anyway to get assigned to an issue. It looks bad when you worked and someone else pushed a PR.

haixuanTao, 2025-03-07:
The thing is that I don't really know the people who ask to get assign on the issue.

If you are afraid of working on something someone is already working on. I'm going to raise additional first issues so that it works for everyone.

MunishMummadi, 2025-03-07:
I completely understand what you are saying. I live in a different timezone so. No problem. I could have pushed a PR.

MunishMummadi, 2025-03-07:
I looked it up and started working on it but it was assigned to someone else. No worries, I will try to stay on watch. I made an issue care to drop your views #822

haixuanTao, 2025-03-16:
Closed as of #812

closed 2025-03-16 by haixuanTao