dora-rs/dora #814: Adding Phi-4 multimodal

haixuanTao, 2025-03-07:
In order to improve audio performance, it could be good to add a dora-phi4 node that can read from audio, text and image by replicating nodes such as https://github.com/dora-rs/dora/blob/main/node-hub/dora-qwen2-5-vl

The model: https://huggingface.co/microsoft/Phi-4-multimodal-instruct

To get started on a new node, you can just:

```bash
dora new dora-phi4 --lang python --type node
```

```bash
cd dora-phi4
uv pip install -e .
```

7SOMAY, 2025-03-07:
Hi @haixuanTao, I guess command should be this : dora new dora-phi4 --lang python --kind node (instead of --type)

haixuanTao, 2025-03-07:
Indeed sorry!

7SOMAY, 2025-03-07:
Please assign this issue to me.

haixuanTao, 2025-03-07:
Done :)

haixuanTao, 2025-03-16:
Closed as of #861

closed 2025-03-16 by haixuanTao