dora-rs/dora #816: Add a dora-llama-cpp-python node

haixuanTao, 2025-03-07:
In order to support more model without having to build additional nodes every time for each model. It could be great to have a `dora-llama-cpp-python` that is just https://github.com/dora-rs/dora/blob/main/node-hub/dora-qwen without the additional backend and with the model made configurable based on environment variables. 

Ideally, this node would also support images, and historical data.

haixuanTao, 2025-03-07:
An example of a working pyproject.toml

```toml
[project]
name = "dora-qwen"
version = "0.3.10"
authors = [{ name = "Your Name", email = "email@email.com" }]
description = "dora-qwen"
license = { text = "MIT" }
readme = "README.md"
requires-python = ">=3.9"

dependencies = [
    "dora-rs >= 0.3.9",
    "torch == 2.4.0",
    "torchvision >= 0.19",
    "torchaudio >= 2.1.0",
    "opencv-python >= 4.1.1",
    "modelscope >= 1.18.1",
    "accelerate>=1.3.0",
    "transformers",
    "mlx-lm>=0.21.1; sys_platform == 'darwin'",
    "llama-cpp-python",
]

[tool.uv.sources]
llama-cpp-python = [
    { index = "llama_cpp_python_metal", marker = "sys_platform == 'darwin'" },
    { index = "llama_cpp_python_cu121", marker = "sys_platform == 'linux'" },
]

[dependency-groups]
dev = ["pytest >=8.1.1", "ruff >=0.9.1"]

[[tool.uv.index]]
name = "llama_cpp_python_cu121"
url = "https://abetlen.github.io/llama-cpp-python/whl/cu121"
explicit = true

[[tool.uv.index]]
name = "llama_cpp_python_metal"
url = "https://abetlen.github.io/llama-cpp-python/whl/metal"
explicit = true

[project.scripts]
dora-qwen = "dora_qwen.main:main"
```

Note that there is different index based on the backend

ShashwatPatil, 2025-03-07:
please assign me

haixuanTao, 2025-03-07:
Done :)

ShashwatPatil, 2025-03-11:
I have made the pull request #850 ,
Please review it @haixuanTao

haixuanTao, 2025-03-16:
Closed as of #850

closed 2025-03-16 by haixuanTao