dora-rs/dora #817: Add a dora-lmdeploy node

haixuanTao, 2025-03-07:
LmDeploy showcase great performance on inference on Cuda and it could be great to wrap a node called dora-lmdeploy that could specify the model we want to use and handle the logic around receiving text and images. 

It would probably be very similar to https://github.com/dora-rs/dora/blob/main/node-hub/dora-qwen2-5-vl except that we replace transformers with lmdeploy pipelines.

7SOMAY, 2025-03-07:
Please assign this issue to me!

Shar-jeel-Sajid, 2025-03-12:
@dora-bot assign me

7SOMAY, 2025-03-13:
@dora-bot assign me

Shar-jeel-Sajid, 2025-03-16:
@dora-bot unassign me

MunishMummadi, 2025-03-17:
@7SOMAY Can I work on this issue? Since I had worked on vl models. I think i can pick this up!!

7SOMAY, 2025-03-17:
@dora-bot unassign me

MunishMummadi, 2025-03-17:
@dora-bot assign me

MunishMummadi, 2025-03-19:
@dora-bot unassign me

7SOMAY, 2025-03-19:
@dora-bot assign me

akshay-juneja, 2025-09-06:
@dora-bot assign me