dora-rs/dora #822: Create Dora-bot for issue assignment and unassignment

MunishMummadi, 2025-03-07:
**Is your feature request related to a problem? Please describe.**
- We need to have/create a gh bot that will automatically assign and unassign users to issues. 

**Describe the solution you'd like**
I suggest have a gh bot(dora-bot it's cute) that will allow users to do the following:
- Assigning themselves to an issue just by commenting @dora-bot assign me
- Unassigning themselves by commenting @dora-bot unassign me
- A message as they are assigned or unassigned

This solution can be implemented by gh Actions with a workflow file that will examine the issue comments and do the appropriate assignment.

**Describe alternatives you've considered**
- An alternative is maintainer will manually assign issues to users.
- Third party tools(will dig deep if needed)

**Additional context**
- This will help users to assign themselves and unassign themselves creating an autonomous process.
- I am happy to push a patch. If this is issue is accepted.

MunishMummadi, 2025-03-07:
@haixuanTao would love your opinion

yogeshsingh360, 2025-03-08:
I am found solution for this issue; can you assign me it to me.

MunishMummadi, 2025-03-08:
I am a GSOC contributor. I created this issue so that I can work on this one!!

haixuanTao, 2025-03-09:
Yes okay @MunishMummadi You can work on a dora-bot :) Sorry for the slow response.

haixuanTao, 2025-03-09:
I have opened two additional issues: #828  #829  that I think should be accessible for many.

MunishMummadi, 2025-03-11:
Hey @haixuanTao Based on the logs from `Actions` : https://github.com/dora-rs/dora/actions/runs/13789953859/job/38566950952 
- `"message": "Resource not accessible by integration"`. This means the GITHUB_TOKEN that's automatically provided to your workflow does not have the necessary permissions to modify the issue (add assignees and comments).
- I can use a gh cli based approach like using gh graphql api.
- If you have any suggestions. Please let me know. I am happy to make a new PR

phil-opp, 2025-03-11:
Maybe this is enough to fix this: https://github.com/dora-rs/dora/pull/848

@MunishMummadi I think the `gh` cli is installed by default on the CI runners. So we should be able to simplify the CI script you added in #840. For example, you should be able to assign someone to an issue by running `gh issue edit <issue-number> --add-assignee <github-username>`.

phil-opp, 2025-03-11:
@dora-bot assign me

phil-opp, 2025-03-11:
Seems to work now after merging #848 🚀 

By the way, who of us has access to the @dora-bot GitHub handle? Or are we pinging a random unrelated GitHub user every time?

MunishMummadi, 2025-03-11:
> Maybe this is enough to fix this: [#848](https://github.com/dora-rs/dora/pull/848)
>
> [@MunishMummadi](https://github.com/MunishMummadi) I think the `gh` cli is installed by default on the CI runners. [...]

Thanks for this!! I would have complicated it :)

MunishMummadi, 2025-03-11:
> Seems to work now after merging [#848](https://github.com/dora-rs/dora/pull/848) 🚀
> 
> By the way, who of us has access to the [@dora-bot](https://github.com/dora-bot) GitHub handle? Or are we pinging a random unrelated GitHub user every time?

Since I have used a hardcoded approach. We will be pinging a random `gh` user haha. I hope it won't be problem. If it is needed, I can create a full pledged solution.

haixuanTao, 2025-03-16:
Closed as of #848