dora-rs/dora #829: Use pyo3 experimental-inspect instead of `generate_stubs.py`

haixuanTao, 2025-03-09:
Currently dora use a custom script called `generate_stubs.py` to generate type hinting in python.

Unfortunately, this typing is not perfect as it requires manual type hinting and has many depreciated feature when linted with:

```bash
uvx ruff check --select UP
```

We can try to run pyo3 with the feature flag experimental-inspect and see if this can be a good automatic stub generator. 

To do this, we just need to replace in `Cargo.toml`:

```toml
pyo3 = { version = "0.23", features = [
    "eyre",
    "abi3-py37",
    "multiple-pymethods",
] }
```

with:

```toml
pyo3 = { version = "0.23", features = [
    "eyre",
    "abi3-py37",
    "multiple-pymethods",
    "experimental-inspect"
] }
```

Ignavar, 2025-03-10:
Hey, I was just working on this issue. Could you clarify how I can check if the issue is resolved after making the changes? That would be really helpful in troubleshooting.

Ignavar, 2025-03-12:
I believe it's still not working. I made the necessary changes and followed the steps you provided, but I’m still encountering an error when inspecting it.
[image]

haixuanTao, 2025-03-12:
OK thanks for trying :S

Krishnadubey1008, 2025-03-15:
@haixuanTao I have been thinking that after adding "experimental-inspect" to `cargo.toml` and removing `generate_stub.py` should I need to make changes in code of `binaries/cli/src/lib.rs`(or any other file). Please guide me!

haixuanTao, 2025-03-15:
No, the `experimental-inspect` is only for dora-node-api-python.

Jayeshs27, 2025-03-29:
@haixuanTao 
- I Added the experimental-inspect feature to PyO3 in `Cargo.toml`:
```
pyo3 = { version = "0.23", features = [
    "eyre",
    "abi3-py37",
    "multiple-pymethods",
    "experimental-inspect"
] }
```
- and `async = ["pyo3/experimental-async", "pyo3/experimental-inspect"]` in `apis/python/node`
- Built the project with this feature enabled
- Created a test script to check if type annotations were accessible through Python's inspection mechanisms
- Attempted to generate improved stubs using the type information

### Conclusions
- The experimental-inspect feature doesn't appear to expose type annotations through Python's standard inspection mechanisms
- No `__pyo3_annotations__` attributes were found on any classes or functions
- All classes had empty __annotations__ dictionaries
- The generated stubs still lack proper type information

```
Class: DoraStatus
  Class annotations: {}

Class: Enum
  Class annotations: {}
  Method: __new__
    Signature: (cls, value)
    Annotations: {}

Class: Node
  Class annotations: {}

Class: Ros2Context
  Class annotations: {}

Class: Ros2Durability
  Class annotations: {}

Class: Ros2Liveliness
  Class annotations: {}

Class: Ros2Node
  Class annotations: {}

Class: Ros2NodeOptions
  Class annotations: {}

Class: Ros2Publisher
  Class annotations: {}

Class: Ros2QosPolicies
  Class annotations: {}

Class: Ros2Subscription
  Class annotations: {}

Class: Ros2Topic
  Class annotations: {}
```

haixuanTao, 2025-03-30:
Ah okay. it's bummer. Thanks for this anyway