dora-rs/dora #830: Create a dora-transformers node that can handle multiple model and modality!

haixuanTao, 2025-03-09:
In order to facilitate integration of multiple AI models, I think it could be good to have a generic dora-transformers node that can be used to try different AI models.

I think that we can reuse most of the work already made on dora-qwen2.5vl and build on top of it.

Mati-ur-rehman-017, 2025-03-12:
@dora-bot assign me

Krishnadubey1008, 2025-03-14:
@Mati-ur-rehman-017 are you still working on this issue

Mati-ur-rehman-017, 2025-03-14:
Yes still working  on the issue. How do i test a node completely through different inputs?  I am currently changing models based on environment variables and to test these models i have to use different inputs. Currently i plan to run python scripts separately for testing. is there a more efficient method? @haixuanTao

MunishMummadi, 2025-03-14:
Hey, I am just curious. what do you mean different inputs? Are u planning to pass image inputs??(for ex: dora-qwen2.5l) to see the output??. I believe as long as your model is up and running and you are following the chat template [example](https://github.com/dora-rs/dora/blob/main/node-hub/dora-qwen2-5-vl/dora_qwen2_5_vl/chat_template.json) or various chat template/ways you should be good to go. Please correct me if i am wrong!!

haixuanTao, 2025-03-14:
Ok going to work on this this weekend

MunishMummadi, 2025-03-14:
Can I lend a hand? If it's ok!! multi model modality sounds interesting.

Mati-ur-rehman-017, 2025-03-15:
@dora-bot unassign me

ShashwatPatil, 2025-03-15:
Hey, while developing the dora-llama node I had previously made the node using transformers along with support for multiple models,
@haixuanTao can I use that for this transformer node, or is there something else needed in this node?

haixuanTao, 2025-03-15:
Yes, you can just use the transformers code and just remove all support for dora-llama-cpp-python making them strictly separated.

haixuanTao, 2025-03-16:
> Ok going to work on this this weekend

So I have added a minimal testing dataflow in https://github.com/dora-rs/dora/blob/main/tests/llm/phi4.yaml that we can have a look and test very simple prompt and check for the response.

monickverma, 2025-03-27:
can i work on the issue?

iamalok123, 2025-05-09:
assign me