dora-rs/dora #839: Add dora-magma node

haixuanTao, 2025-03-10:
Add support for Microsoft Magma model: https://github.com/microsoft/Magma?tab=readme-ov-file

This would require to add magma as a git submodule within the nodes in the likes of dora-rdt-1b and then connect the model from transformers in the likes of: https://github.com/microsoft/Magma/blob/main/agents/robot_traj/app.py

Making sure the input and output matches requirements.

Note that it seems that this uses a different transformers model.

MunishMummadi, 2025-03-11:
I might need little guidance with this issue!!

MunishMummadi, 2025-03-11:
@dora-bot assign me

MunishMummadi, 2025-03-11:
@dora-bot assign me

MunishMummadi, 2025-03-11:
@dora-bot assign me

Shar-jeel-Sajid, 2025-03-12:
@dora-bot assign me

Codexgigaa, 2025-03-12:
It looks like this task has been pending for over 19 hours without updates. If no progress has been made, I’d like to take over and proceed with the implementation. Please let me know if I can work on this. Thanks!

MunishMummadi, 2025-03-12:
Thank you for your concern. Last i checked/wrote it takes 14 days for an issue to be considered stale/no-activity. I write my code — so, it will take some time sorry.

MunishMummadi, 2025-03-12:
Update: I am testing few changes. I will be able to push a patch in few hours.

Codexgigaa, 2025-03-12:
Thanks for the update! I’ll start working on creating the dora-magma-recorder to ensure data is collected in the correct format for training and fine-tuning Magma. I'll first review the default repo to understand the expected data structure and then integrate the recorder within Dora. Let me know if there are any specific requirements or constraints to consider.

haixuanTao, 2025-03-16:
Closed as of #863

closed 2025-03-16 by haixuanTao