dora-rs/dora #852: Improve IntoArrow within our arrow-convert crate

haixuanTao, 2025-03-12:
Currently we implement automatic rust conversion from simple types to arrow but we still miss a lot of types.

We could greatly benefit from having more automatic into arrow and from arrow conversion in dora-rs.

The file to modify is: https://github.com/dora-rs/dora/blob/c6ce45785fa1b130030ec233eebb9d463c5b8664/libraries/arrow-convert/src/into_impls.rs
and: https://github.com/dora-rs/dora/blob/52045f3af84924e5a1cbc3283ade98a962514089/libraries/arrow-convert/src/from_impls.rs

This should be a good first issue for rust dev as this is fully contained within a simple crate.

Tushar4059x, 2025-03-13:
@haixuanTao What are the highest priority missing types we should implement first?

haixuanTao, 2025-03-13:
- Borrowed list
- string
- boolean
- temporal

We can take inspiration from: https://docs.rs/arrow_convert/latest/arrow_convert/

I wonder if we can even import it and use it as is!

Shar-jeel-Sajid, 2025-03-13:
I have read the [https://docs.rs/arrow_convert/latest/arrow_convert/](https://docs.rs/arrow_convert/latest/arrow_convert/) and i think we will be able to import and use it as it is. It has default support for

1. Numeric types
2. Bools
3. String
4. Temporal
and also allows to do other conversions. 
Should i try to import it in dora?

Shar-jeel-Sajid, 2025-03-13:
Also Do we have test cases to test the round trip conversion from rust to arrow and arrow to rust?

Shar-jeel-Sajid, 2025-03-14:
I read about [https://docs.rs/arrow_convert/latest/arrow_convert/](https://docs.rs/arrow_convert/latest/arrow_convert/). I think we can import and integrate it in dora. Should i use this crate?

haixuanTao, 2025-03-14:
We can! If we can make it not break the current API

Adarshgupta1127, 2025-03-28:
@dora-bot assign this issue to me

haixuanTao, 2025-04-13:
Close as we implement most of them as of #864

closed 2025-04-13 by haixuanTao