dora-rs/dora #854: Improve C++ Arrow type Conversion

haixuanTao, 2025-03-13:
We should improve dora C++ arrow type conversion at https://github.com/dora-rs/dora/blob/2488d0916c6dac4b6acb404d600c2ed025fd62d1/apis/c++/node/src/lib.rs

We can definetely do more than:
```c++
    let data = match metadata.type_info.data_type {
        dora_node_api::arrow::datatypes::DataType::UInt8 => {
            let array: &UInt8Array = data.as_primitive();
            array.values().to_vec()
        }
        dora_node_api::arrow::datatypes::DataType::Null => {
            vec![]
        }
        _ => {
            todo!("dora C++ Node does not yet support higher level type of arrow. Only UInt8. 
                The ultimate solution should be based on arrow FFI interface. Feel free to contribute :)")
        }
    };

```

Codexgigaa, 2025-03-13:
@dora-bot assign me

Codexgigaa, 2025-03-13:
Hello @haixuanTao ,

I hope you're doing well! I have a few doubts regarding the project and would really appreciate your guidance. Could you please let me know the best place or **platform where I can ask my questions** or discuss these doubts with you?

Thank you for your help!

Best regards,
Lovish Bhardwaj

haixuanTao, 2025-03-13:
We use discord: https://discord.gg/QHA96fP3

Mati-ur-rehman-017, 2025-03-16:
In c++, currently we are sending raw databytes to rust function which are converted to arrow in rust and then system calls are made. why dont we convert to arrow in cpp and then make rust function calls on this formatted data directly. Converting back to any language format will be easy due to arrow support for languages and datatypes. Please guide @haixuanTao

Mati-ur-rehman-017, 2025-03-16:
We can provide function similar to [https://github.com/dora-rs/dora/blob/2488d0916c6dac4b6acb404d600c2ed025fd62d1/apis/c%2B%2B/node/src/lib.rs](https://github.com/dora-rs/dora/blob/2488d0916c6dac4b6acb404d600c2ed025fd62d1/apis/c%2B%2B/node/src/lib.rs) for conversion to arrow.

haixuanTao, 2025-03-16:
Yeah absolutely, we should probably use something like arrow.cpp and accept.a arrow.cpp pointer within rust deserialize it and then send the data. Same for the serialization part. 

But, as far as I know, I had trouble finding a good arrow.cpp "client". But that could definitely be a good route.

An other route is also to convert into Builtins Type by default given the right function call: https://github.com/dtolnay/cxx?tab=readme-ov-file#builtin-types making it possibly easier for non-arrow C++ users.

FYI, I'm in no mean a C++ expert and so this is purely speculative,

ankitmrj, 2025-03-23:
Hi @haixuanTao,

Thank you for sharing your thoughts! I agree that leveraging arrow.cpp for Arrow data handling in C++ and passing pointers to Rust for deserialization is a great approach. It would indeed improve efficiency and interoperability between C++ and Rust. However, I understand your concern about finding a reliable arrow.cpp client. Here are a few suggestions to address this:

1.Alternative: Built-in Types for Simplicity
For users who don’t need Arrow-specific functionality, we can provide an option to convert data into built-in types (e.g., std::vector, std::string) using the cxx crate. This would make the API more accessible to non-arrow C++ users.
C++
#include <vector>
#include <string>

extern "C" {
    void rust_process_vector(const std::vector<uint8_t>& data);
    void rust_process_string(const std::string& data);
}

void send_data_to_rust() {
    std::vector<uint8_t> data = {1, 2, 3};
    rust_process_vector(data);

    std::string message = "Hello, Rust!";
    rust_process_string(message);
}
Rust
#[cxx::bridge]
mod ffi {
    extern "C++" {
        include!("path/to/cpp/header.h");
        fn rust_process_vector(data: &Vec<u8>);
        fn rust_process_string(data: &String);
    }
}

fn main() {
    let data = vec![1, 2, 3];
    ffi::rust_process_vector(&data);

    let message = String::from("Hello, Rust!");
    ffi::rust_process_string(&message);
}
I can start by implementing support for more Arrow data types (e.g., Int32, Float64, String) in the Rust match statement.

Then, I can work on integrating the Arrow C++ library and FFI for efficient data exchange.

Finally, I can add helper functions and tests to ensure the implementation is robust and user-friendly.

Let me know if this approach sounds good, or if you have any additional suggestions!

Best regards,
Ankit Srivastav

haixuanTao, 2025-03-24:
I think ultimately we need to support both, so indeed could be good to work on the native support first.

Thanks :)

haixuanTao, 2025-04-08:
Merged as of #918

closed 2025-04-08 by haixuanTao