dora-rs/dora #865: failed to compile `dora-cli v0.3.10`

451846939, 2025-03-15:
**Describe the bug**
error: could not compile `zenoh` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: failed to compile `dora-cli v0.3.10`

error[E0609]: no field `flow` on type `DownsamplingItemConf`
  --> /Users/xxxx/.cargo/registry/src/rsproxy.cn-0dccff568467c15b/zenoh-1.2.1/src/net/routing/interceptor/downsampling.rs:59:24
   |
59 |             flow: conf.flow,
   |                        ^^^^ unknown field
   |
help: a field with a similar name exists
   |
59 |             flow: conf.flows,
   |                        ~~~~~

This issue may need to be resolved as the installation fails.
https://github.com/eclipse-zenoh/zenoh/issues/1833

**To Reproduce**
Steps to reproduce the behavior:
cargo install dora-cli

451846939, 2025-03-15:
Running cargo install dora-cli --locked can resolve the issue.

451846939, 2025-03-15:
dora-daemon

451846939, 2025-03-15:
Cargo.lock uses zenoh 1.2.1, so it may need to be modified.

[[package]]
name = "dora-daemon"
version = "0.3.10"
dependencies = [
 "aligned-vec",
 "async-trait",
 "bincode",
 "crossbeam",
 "crossbeam-skiplist",
 "ctrlc",
 "dora-arrow-convert",
 "dora-core",
 "dora-download",
 "dora-message",
 "dora-node-api",
 "dora-tracing",
 "eyre",
 "flume 0.10.14",
 "futures",
 "futures-concurrency",
 "serde_json",
 "serde_yaml 0.8.26",
 "shared-memory-server",
 "sysinfo 0.30.13",
 "tokio",
 "tokio-stream",
 "tracing",
 "tracing-opentelemetry",
 "uuid",
 "which",
 "zenoh 1.2.1",
]

Choudhry18, 2025-03-15:
@dora-bot assign me

phil-opp, 2025-03-16:
> Cargo.lock uses zenoh 1.2.1, so it may need to be modified.

Using zenoh 1.2.1 is not the issue, the problem is that there is an incompatible 1.2.2 release of `zenoh-config`. Our lockfile still uses the compatible 1.2.1 release of `zenoh-config`, so there is nothing that needs to be done. You are required to use `cargo install --locked` however until they fix it upstream.

451846939, 2025-03-16:
Thank you for your reply. I will close this issue.

closed 2025-03-16 by 451846939