dora-rs/dora #878: Tracking Issues for better code quality flag in Python

haixuanTao, 2025-03-17:
Follow up flag that can easily be added:

- [ ] Use 'I' for isort imports
- [ ] Use 'RET' for flake8 Return
- [ ] Use 'RSE' for flake8 Raise
- [ ] Use "E" for pycodestyle. Line too long warning. Requires an automatic comment line reducers
- [ ] Use "PERF" for performance
- [ ] Use "NPY" for numpy related errors
- [ ] Use "N" for better naming

haixuanTao, 2025-03-17:
Each Letter can be assigned individually.

7SOMAY, 2025-03-17:
@dora-bot, assign me

haixuanTao, 2025-03-17:
@7SOMAY  Could you say which letter you want to tackle next as to let other people do other ones?

7SOMAY, 2025-03-17:
Can I work on more than one letters?

7SOMAY, 2025-03-17:
@haixuanTao, please assign "PERF" to me :)

Ignavar, 2025-03-17:
@haixuanTao please assign "I" to me

MunishMummadi, 2025-03-17:
@haixuanTao I would like to work on "NPY"

7SOMAY, 2025-03-17:
@haixuanTao, please assign me one more : “RSE”

haixuanTao, 2025-03-18:
@7SOMAY this devil has fixed many of the current flags. 

Verry sorry for the people that wanted to work on this

7SOMAY, 2025-03-19:
@dora-bot assign me