dora-rs/dora #885: Improve Users-Getting Started on dora-rs.ai

haixuanTao, 2025-03-17:
The current getting started for dora-rs is still a bit missing a lot of components to make it workable by anyone,

A thorough rewrite in order to split concept could really improve users getting started with dora-rs.

Shiv-aurora, 2025-03-17:
@dora-bot assign me

Shiv-aurora, 2025-03-18:
@Krishnadubey1008 hey I'm already working on that, so i would apreacitate if you could find another issue to contribute.

Krishnadubey1008, 2025-03-18:
@Shiv-aurora as I had edited only "Python Conversation" file rest 3 files you can work on

Krishnadubey1008, 2025-03-18:
@haixuanTao while running the dataflow using instructions in [https://dora-rs.ai/docs/guides/getting-started/conversation_py](url) I had been encountering issue-ImportError: cannot import name 'Node' from 'dora'

[image]
This was because virtual enviroment was not activated before running commands(such as pip install dora-rs-cli). When I did that I got the correct output 

[image]

Please review my PR [https://github.com/dora-rs/dora-rs.github.io/pull/27](url) .

Shiv-aurora, 2025-03-19:
Hi @haixuanTao 

I’ve submitted Part 1 of 4 for this issue in PR  [dora-rs/dora-rs.github.io#28](https://github.com/dora-rs/dora-rs.github.io/pull/28) 
Please review it and let me know if you have any feedback before I proceed with the remaining parts!

kingchou007, 2025-06-16:
@dora-bot assign me

LIKHITHA-NAGRAJ, 2026-03-01:
@dora-bot assign me