dora-rs/dora #889: [Bug] Failed to run dora-daemon due to UDP multicast binding error in WSL

chen-cdd, 2025-03-18:
### Environment
- OS: Windows WSL (Ubuntu)
- Dora Version: 0.3.10
- Rust Version: 1.85.0

### Problem Description
When trying to run `dora up` in WSL environment, the daemon fails to start due to UDP multicast binding issues.

### Error Message
```
[ERROR]
failed to run dora-daemon

Caused by:
   0: failed to open zenoh session
   1: Unable to bind UDP port 224.0.0.224:7446 at /home/rooter/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/zenoh-1.2.1/src/net/runtime/orchestrator.rs:649. - Caused by Cannot assign requested address (os error 99)
   2: Cannot assign requested address (os error 99)

Location:
    /home/rooter/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/dora-daemon-0.3.10/src/lib.rs:266:2
```

Hennzau, 2025-03-18:
Hi @chen-cdd, I know that UDP multicast is not enabled on windows by default, do you think it could be the source of this error?

haixuanTao, 2025-03-18:
Follow up, this is only WSL right? Normal Windows work?

chen-cdd, 2025-03-19:
> Follow up, this is only WSL right? Normal Windows work?

yes,only wsl and windows is normal😢and i  have already enabled this and added the port, but my current error is as follows, and it seems like it still hasn’t been resolved:

(.venv) rooter@chen:/mnt/f/AAAhackthon/mofa/mofa-main/python/examples/hello_world$ dora up
started dora daemon

[ERROR]
failed to run dora-daemon

Caused by:
   0: failed to open zenoh session
   1: Unable to bind UDP port 224.0.0.224:7446 at /home/rooter/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/zenoh-1.2.1/src/net/runtime/orchestrator.rs:649. - Caused by Cannot assign requested address (os error 99)
   2: Cannot assign requested address (os error 99)

Location:
    /home/rooter/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/dora-daemon-0.3.10/src/lib.rs:266:26

chen-cdd, 2025-03-19:
> Hi [@chen-cdd](https://github.com/chen-cdd), I know that UDP multicast is not enabled on windows by default, do you think it could be the source of this error?

I have already enabled this and added the port, but my current error is as follows, and it seems like it still hasn’t been resolved:

(.venv) rooter@chen:/mnt/f/AAAhackthon/mofa/mofa-main/python/examples/hello_world$ dora up
started dora daemon

[ERROR]
failed to run dora-daemon

Caused by:
   0: failed to open zenoh session
   1: Unable to bind UDP port 224.0.0.224:7446 at /home/rooter/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/zenoh-1.2.1/src/net/runtime/orchestrator.rs:649. - Caused by Cannot assign requested address (os error 99)
   2: Cannot assign requested address (os error 99)

Location:
    /home/rooter/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/dora-daemon-0.3.10/src/lib.rs:266:26

starlitxiling, 2025-03-19:
I think it has nothing to do with windows and wsl, I can run `dora up` correctly on windows and wsl, can you check if the port is occupied?

chen-cdd, 2025-03-19:
> I think it has nothing to do with windows and wsl, I can run `dora up` correctly on windows and wsl, can you check if the port is occupied?

There were no ports occupied, then I reinstalled dora, cargo install dora-cli --locked, and now the error is reported as follows：
(.venv) rooter@chen:/mnt/f/AAAhackthon/mofa/mofa-main/python/examples/hello_world$ dora up
started dora coordinator

[ERROR]
failed to run dora-coordinator

Caused by:
   0: failed to create local TCP listener
   1: No such file or directory (os error 2)

Location:
    /home/rooter/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/dora-coordinator-0.3.10/src/listener.rs:15:24

starlitxiling, 2025-03-19:
Can you tell me your wsl version? Starting from WSL 2.0.9, Windows Firewall rules are automatically applied to WSL. You can refer to this link https://learn.microsoft.com/en-us/windows/wsl/networking. I think you need to enable Dora app to access Windows private network and public network in the firewall

chen-cdd, 2025-03-19:
> Can you tell me your wsl version? Starting from WSL 2.0.9, Windows Firewall rules are automatically applied to WSL. You can refer to this link [...]

成功运行了！！（虽然不知道为什么！！但是感谢你们🌹

chen-cdd, 2025-03-19:
成功运行了，在操作你们给我的建议后，重启电脑就成功了，感谢你们🌹🌹🌹

starlitxiling, 2025-03-19:
ok, I think this issue can be closed.

haixuanTao, 2025-03-19:
So this was solved by updating WSL to version 2.0