dora-rs/dora #890: Create generic arrow primitive casting for user to not have to implement a function for each arrow type

haixuanTao, 2025-03-18:
Currently within rust, we need to match all possible arrow type for each input which can be very cumbersome.

If we could instead have either a procedural macro or a generic function that can deserialize all similar type into the one we want it would be easier for users. 

For example within dora-rerun, We currently have the following expression:

```rust
dora_node_api::arrow::datatypes::DataType::Float32 => {
            let buffer: &Float32Array = data
                .as_any()
                .downcast_ref()
                .context("series is not float32")?;
            let series: Vec<_> = buffer.values().to_vec();
            for (i, value) in series.iter().enumerate() {
                rec.log(
                    format!("{}_{}", id.as_str(), i),
                    &rerun::Scalar::new(*value as f64),
                )
                .wrap_err("could not log series")?;
            }
        }
        dora_node_api::arrow::datatypes::DataType::Float64 => {
            let buffer: &Float64Array = data
                .as_any()
                .downcast_ref()
                .context("series is not float64")?;
            let series: Vec<_> = buffer.values().to_vec();
            for (i, value) in series.iter().enumerate() {
                rec.log(
                    format!("{}_{}", id.as_str(), i),
                    &rerun::Scalar::new(*value),
                )
                .wrap_err("could not log series")?;
            }
        }
        dora_node_api::arrow::datatypes::DataType::Int32 => {
            let buffer: &Int32Array = data
                .as_any()
                .downcast_ref()
                .context("series is not Int32")?;
            let series: Vec<_> = buffer.values().to_vec();
            for (i, value) in series.iter().enumerate() {
                rec.log(
                    format!("{}_{}", id.as_str(), i),
                    &rerun::Scalar::new(*value as f64),
                )
                .wrap_err("could not log series")?;
            }
        }
        dora_node_api::arrow::datatypes::DataType::Int64 => {
            let buffer: &Int64Array = data
                .as_any()
                .downcast_ref()
                .context("series is not Int64")?;
            let series: Vec<_> = buffer.values().to_vec();
            for (i, value) in series.iter().enumerate() {
                rec.log(
                    format!("{}_{}", id.as_str(), i),
                    &rerun::Scalar::new(*value as f64),
                )
                .wrap_err("could not log series")?;
            }
```

This is very borrowing as all implementation are essentially the same.

Maybe a simpler more elegant solution for writing the following code can be found.

Krishnadubey1008, 2025-03-19:
@dora-bot assign me

Krishnadubey1008, 2025-03-19:
@haixuanTao I think we can define helper functions as
```
fn process_float32(data: &dyn Array, id: &str, rec: &mut Recorder) -> Result<()> {
    let buffer: &Float32Array = data.as_any().downcast_ref().context("series is not float32")?;
    let series: Vec<_> = buffer.values().to_vec();
    for (i, value) in series.iter().enumerate() {
        rec.log(format!("{}_{}", id, i), &rerun::Scalar::new(*value as f64))
            .wrap_err("could not log series")?;
    }
    Ok(())
}
```
and then Update Macro to Call Helper Functions
```
#[proc_macro]
pub fn generate_match_arms(_input: TokenStream) -> TokenStream {
    let arms = vec![
        quote! { DataType::Float32 => process_float32(data, id, rec), },
        quote! { DataType::Float64 => process_float64(data, id, rec), },
        quote! { DataType::Int32 => process_int32(data, id, rec), },
        quote! { DataType::Int64 => process_int64(data, id, rec), },
    ];

    let expanded = quote! {
        match data_type {
            #(#arms)*
            _ => unimplemented!("Unsupported data type"),
        }
    };

    TokenStream::from(expanded)
}
```

closed 2025-03-20 by haixuanTao