dora-rs/dora #894: Dora should be able to self update itself

Hennzau, 2025-03-19:
As `uv` do, maybe we could extend the CLI so that `dora self update` is updating the CLI to the latest version

What do you think of that?

haixuanTao, 2025-03-19:
Hey Enzo, that's a great idea. And maybe for node as well right?

Maybe we could try with something chill like the uv package manager to help us do this in simple instance could be a good start

Hennzau, 2025-03-19:
Yep we can probably do something for the nodes as well, but for the current workflow, `dora build` should already install any new published version of a node.

I think I will let someone do the `dora self update`, as it should be really easy with https://crates.io/crates/self_update

7SOMAY, 2025-03-19:
@dora-bot assign me

closed 2025-03-25 by haixuanTao