dora-rs/dora #896: Undetailed Documentation for beginners

gagan-1307, 2025-03-19:
The current documentation is complicated to understand and go through the project which requires a lot of hours to get through the codebase and functionality of project.

## Current State
- Basic documentation structure exists
- Some guides and API references are available
- Limited examples and use cases
- Basic installation instructions

## Proposed Improvements to improve readability and understanding

### 1. Getting Started Guide
- [ ] Create a step-by-step tutorial for absolute beginners
- [ ] Add a "Quick Start" guide with minimal setup
- [ ] Include prerequisites and system requirements
- [ ] Add troubleshooting section for common installation issues

### 2. API Documentation
- [ ] Expand Python API documentation with more code examples
- [ ] Add detailed Rust API documentation with practical use cases
- [ ] Include performance considerations and best practices
- [ ] Add type hints and return value documentation

### 3. Architecture and Concepts
- [ ] Create detailed architecture diagrams
- [ ] Explain core concepts with visual aids
- [ ] Document the data flow between nodes
- [ ] Add system design principles and patterns

### 4. Examples and Tutorials
- [ ] Add more real-world use cases
- [ ] Create tutorials for common scenarios
- [ ] Include complete end-to-end examples
- [ ] Add performance optimization examples

### 5. Troubleshooting Guide
- [ ] Create a comprehensive troubleshooting section
- [ ] Add common error messages and solutions
- [ ] Include debugging tips and tools
- [ ] Add performance tuning guidelines

haixuanTao, 2025-03-19:
Absolutely agree that the documentation is not great! 

Please help us out!

gagan-1307, 2025-03-19:
@haixuanTao should i work on documentation and include the changes.

Hennzau, 2025-03-19:
> [@haixuanTao](https://github.com/haixuanTao) should i work on documentation and include the changes.

It would be really cool!

vedantvijay, 2025-03-29:
@dora-bot assign me

Xlqmu, 2025-05-15:
I think Deep_Wiki can be used to assist in building Dora's documentation.