dora-rs/dora #899: RuntimeError: Could not initiate node from environment variable. For dynamic node, please add a node id in the initialization function.

G1017, 2025-03-20:
运行https://dora-rs.ai/zh-CN/docs/guides/getting-started/llm中的llm遇到一下问题 请问怎么解决

Traceback (most recent call last):
  File "\\?\C:\Users\yang.geng\PycharmProjects\metahuman\robet\LLMs\keyboard_op.py", line 7, in <module>
    node = Node()
RuntimeError: Could not initiate node from environment variable. For dynamic node, please add a node id in the initialization function.

Caused by:
   0: failed to init event stream
   1: subscribe failed: Node object_detection exited before initializing dora. For more information, run `dora logs 0195b27b-2ff4-7579-a310-0fcadd989b81 object_detection`.

Location:
    apis\rust\node\src\event_stream\mod.rs:139:17

G1017, 2025-03-20:
(dora_env) C:\Users\yang.geng\PycharmProjects\metahuman\robet\LLMs>conda info

     active environment : dora_env
    active env location : C:\Users\yang.geng\AppData\Local\anaconda3\envs\dora_env
            shell level : 2
       user config file : C:\Users\yang.geng\.condarc
 populated config files : C:\Users\yang.geng\.condarc
          conda version : 23.3.1
    conda-build version : 3.24.0
         python version : 3.10.9.final.0
       virtual packages : __archspec=1=x86_64
                          __win=0=0
       base environment : C:\Users\yang.geng\AppData\Local\anaconda3  (writable)
      conda av data dir : C:\Users\yang.geng\AppData\Local\anaconda3\etc\conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\yang.geng\AppData\Local\anaconda3\pkgs
                          C:\Users\yang.geng\.conda\pkgs
                          C:\Users\yang.geng\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\yang.geng\AppData\Local\anaconda3\envs
                          C:\Users\yang.geng\.conda\envs
                          C:\Users\yang.geng\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.32.3 CPython/3.10.9 Windows/10 Windows/10.0.19044
          administrator : False
             netrc file : None
           offline mode : False

G1017, 2025-03-20:
(dora_env) C:\Users\yang.geng\PycharmProjects\metahuman\robet\LLMs>pip list
Package               Version
--------------------- -----------
accelerate            1.5.2
aiohappyeyeballs      2.6.1
aiohttp               3.11.14
aiosignal             1.3.2
annotated-types       0.7.0
anyio                 4.9.0
async-timeout         5.0.1
attrs                 25.3.0
auto_gptq             0.7.1
certifi               2025.1.31
cffi                  1.17.1
charset-normalizer    3.4.1
colorama              0.4.6
datasets              3.4.1
dill                  0.3.8
distro                1.9.0
dora-rs               0.3.10
dora-rs-cli           0.3.10
exceptiongroup        1.2.2
filelock              3.18.0
frozenlist            1.5.0
fsspec                2024.12.0
gekko                 1.2.1
h11                   0.14.0
httpcore              1.0.7
httpx                 0.28.1
huggingface-hub       0.29.3
idna                  3.10
Jinja2                3.1.6
jiter                 0.9.0
joblib                1.4.2
llvmlite              0.44.0
MarkupSafe            3.0.2
more-itertools        10.6.0
mpmath                1.3.0
multidict             6.2.0
multiprocess          0.70.16
networkx              3.4.2
numba                 0.61.0
numpy                 2.1.3
openai                1.67.0
openai-whisper        20240930
optimum               1.24.0
packaging             24.2
pandas                2.2.3
peft                  0.15.0
pillow                11.1.0
pip                   25.0
propcache             0.3.0
psutil                7.0.0
pyarrow               19.0.1
pybind11              2.13.6
pycparser             2.22
pydantic              2.10.6
pydantic_core         2.27.2
pylcs                 0.1.1
pynput                1.8.1
python-dateutil       2.9.0.post0
pytz                  2025.1
PyYAML                6.0.2
regex                 2024.11.6
requests              2.32.3
rouge                 1.0.1
safetensors           0.5.3
scikit-learn          1.6.1
scipy                 1.15.2
sentence-transformers 3.4.1
sentencepiece         0.2.0
setuptools            75.8.0
six                   1.17.0
sniffio               1.3.1
sounddevice           0.5.1
sympy                 1.13.1
threadpoolctl         3.6.0
tiktoken              0.9.0
tokenizers            0.21.1
torch                 2.6.0
tqdm                  4.67.1
transformers          4.49.0
typing_extensions     4.12.2
tzdata                2025.1
urllib3               2.3.0
uv                    0.6.8
wheel                 0.45.1
xxhash                3.5.0
yarl                  1.18.3

G1017, 2025-03-20:
准备工作完成后

G1017, 2025-03-20:
(dora_env) C:\Users\yang.geng\PycharmProjects\metahuman\robet\LLMs>dir
 驱动器 C 中的卷没有标签。
 卷的序列号是 6C10-CEB7

 C:\Users\yang.geng\PycharmProjects\metahuman\robet\LLMs 的目录

2025/03/20  16:58    <DIR>          .
2025/03/20  16:58    <DIR>          ..
2025/03/19  17:34    <DIR>          bge-large-en-v1.5
2025/03/20  15:08             1,706 dataflow.yml
2025/03/19  17:34    <DIR>          deepseek-coder-6.7B-instruct-GPTQ
2025/03/19  17:34             1,188 file_saver_op.py
2025/03/20  16:58             2,662 keyboard_op.py
2025/03/20  14:43             8,788 llm_op.py
2025/03/19  17:34               839 microphone_op.py
2025/03/20  16:58    <DIR>          out
2025/03/19  17:34             2,778 sentence_transformers_op.py
2025/03/20  15:51               608 test.py
2025/03/19  17:34    <DIR>          whisper-base
2025/03/19  17:34               683 whisper_op.py
2025/03/20  14:44    <DIR>          __pycache__

haixuanTao, 2025-03-20:
```python
node = Node("keyboard")
```
Can you modify this line to this?
可以该到这样马？

Hennzau, 2025-03-20:
Hi, I suspect you tried to run directly the `.py` file. To run dora nodes you need to put them in a dataflow description (`YAML` file), and then do `dora run <dataflow.yml>`

closed 2025-03-20 by G1017