dora-rs/dora #921: Add dora-cotracker from facebook/cotracker

haixuanTao, 2025-03-24:
In order to better use models like microsoft/magma it will be necessary to use models such as facebook cotracker: https://github.com/facebookresearch/co-tracker

Could be cool to be able to use in order to track a point for robotic purposes. 

We can get inspired from the demo: 
- https://huggingface.co/spaces/facebook/cotracker
- Code: https://github.com/facebookresearch/co-tracker/blob/main/gradio_demo/app.py

The idea is that from a list of pixel coordinate, we create a query tensor that is tracked with facebook cotracker model:

```python
    model(video_chunk=video_input, is_first_step=True, grid_size=0, queries=queries, add_support_grid=add_support_grid)
````

ShashwatPatil, 2025-03-24:
@dora-bot assign me

Krishnadubey1008, 2025-03-25:
@dora-bot assign me

haixuanTao, 2025-04-07:
Close as implemented at #931

closed 2025-04-07 by haixuanTao