dora-rs/dora #925: Add support for dora-samurai for visual tracking of masks using Samurai!

haixuanTao, 2025-03-26:
The Segment Anything Model 2 (SAM 2)  ( dora-sam2) has demonstrated strong performance in object segmentation tasks but faces challenges in visual object tracking, particularly when managing crowded scenes with fast-moving or self-occluding objects.

dora could greatly benefit from using dora-samurai instead of dora sam2 in order to track masked object in motion.

Choudhry18, 2025-03-26:
@dora-bot assign me

Krishnadubey1008, 2025-03-27:
@haixuanTao do we need to implement a separate directory for `dora-samurai` in `node-hub`

Choudhry18, 2025-03-27:
@Krishnadubey1008 I think the goal is to create a new node in the node hub that implements dora-samurai, the way there is for  dora-sam2. However I am working on the issue and almost done I will probably submit the PR later today, you wanna work on something else?

Choudhry18, 2025-03-28:
@haixuanTao  Apologies for the repeated pings. I’ve finished implementing the node and wanted to test it before submitting the PR.

From my understanding, Samurai relies on an initial bounding box, points, or previous masks for visual tracking. I’m working on an example with a fixed initial bounding box, but I’d like your input on the best approach for determining the initial box for the dora-samurai node when an initial bounding box is not provided.

haixuanTao, 2025-03-28:
Yeah absolutely. I think that in case there is no bounding box, we should not do anything.

Choudhry18, 2025-03-31:
@haixuanTao Thank you for getting back. I had a couple more questions:

1. I have successfully implemented dora-samurai to work with a provided video file or folder of frames. However, dora-sam2 processes a real-time stream of frames and returns masks for each frame. Should dora-samurai also support real-time video inference in a similar way? Currently the samurai implementation doesn't support real-time video inference  - [see](https://github.com/yangchris11/samurai/issues/12) ,but I can try tweaking the package to work if that is our use case. 

2. I was unable to find a pypi package for samurai unlike the one for sam2. https://github.com/dora-rs/dora/blob/4d6fabb59b0f0e8ea926210e8c8d17633ab699e1/node-hub/dora-sam2/pyproject.toml#L14 Adding samurai as a git submodule doesn't work reliability either (they are using hydra and it causes issues with paths when package is used from outside). My workaround was creating a fork of the samurai project and making it a package, I wanted to ask if it is better to keep at as a local package package inside dora-samurai node or publishing it to pypi and use it from there.

ShashwatPatil, 2025-04-04:
Hey @Choudhry18 are you still working on this issue... If not I can look at this issue 🙂️

Choudhry18, 2025-04-04:
@ShashwatPatil I am still working on it, had some delay with real time video inference on samurai but I am almost done now :)