dora-rs/dora #936: Node refuses to shutdown when using `node.next()`

pculbertson-bdai, 2025-04-02:
**Describe the bug**
When using `node.next()` to read messages, the resulting node does not shut down cleanly, and instead gets `SIGKILL`ed after not responding to the shutdown command.

**To Reproduce**
Steps to reproduce the behavior:
```
# spinner.py
import time

from dora import Node

def main() -> None:
    node = Node()

    while True:
        event = node.next()

    # THIS SHUTS DOWN CLEANLY
    # for event in node:
    #     pass

if __name__ == "__main__":
    main()

```
```
# dataflow.yml
nodes:
  - id: spinner
    path: spinner.py
    inputs:
      tick: dora/timer/millis/10
```
`dora run dataflow.yml`
Upon running `ctrl+C`, the workflow will hang, then crash after the 15sec timeout.

**Expected behavior**
The node should shut down cleanly, with the proper `STOP` event / shutdown signal reaching it.

**Environments (please complete the following information):**
 - System info: [use `uname --all` on Linux]
`Linux ws50 6.8.0-52-generic #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 15 19:18:46 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
 - Dora version: [use `dora --version` and `pip show dora-rs`]
`dora-cli 0.3.10`
`Name: dora-rs
Version: 0.3.10
Summary: `dora` goal is to be a low latency, composable, and distributed data flow.
Home-page: 
Author: 
Author-email: 
License: MIT
Location: /home/pculbertson/miniconda3/envs/dexterity/lib/python3.10/site-packages
Requires: pyarrow
Required-by: dora-rs-cli`

eladb, 2025-04-02:
Have you tried to handle to `STOP`?

```py

import time

from dora import Node

def main() -> None:
    node = Node()

    while True:
        event = node.next()

        if event["type"] == "STOP":
          break

if __name__ == "__main__":
    main()
```

phil-opp, 2025-04-02:
The `while True` loop has no break condition, so it never finishes. You should check whether `event` is `None` and `break` in that case.

(Handling the `STOP` event should also work, but you should check whether the returned value is `None` before trying to read the `["type"]` field anyway.)

pculbertson-bdai, 2025-04-02:
Thanks for the quick reply - I was surprised that the SIGINT/SIGKILL signals were not reaching the node, but perhaps this is expected. Handling the `STOP` message explicitly makes this example work.

Except: if the node receives no inputs (e.g., it is a source), the handling fails.
If we amend the example to:
```
# spinner.py
from dora import Node

def main() -> None:
    node = Node()

    while True:
        event = node.next(timeout=1e-3)
        if event is not None and event["type"] == "STOP":
            break

if __name__ == "__main__":
    main()
```
```
# dataflow.yml
nodes:
  - id: spinner
    path: spinner.py
```

The same hanging behavior occurs for me, even with explicit `STOP` message handling.

phil-opp, 2025-04-02:
Dora tries to give nodes a bit of time to stop properly before killing them. The first `ctrl+c` will close the inputs of source nodes and send the stop command. This gives dataflows the opportunity to stop gracefully. If it doesn't stop, we kill it after some timeout.

If you want to kill all nodes directly, you can do a second `ctrl+c`. This should result in an immediate SIGKILL.

> ```
> # dataflow.yml
> nodes:
>   - id: spinner
>     path: spinner.py
> ```
> 
> 
>     
>   
> 
> The same hanging behavior occurs for me, even with explicit `STOP` message handling.

For this node, the event stream should be closed directly. So it _should_ immediately return `None`, but apparently there is some bug. Thanks for reporting this!

(Side note: There is some open discussion on whether we want to special-case nodes without inputs: https://github.com/dora-rs/dora/issues/500 )

pculbertson-bdai, 2025-04-02:
Ah, I think this is actually a duplicate of #500 then - if the node has no inputs, it just hangs since it doesn't receive the shutdown signal. Happy to close this as a duplicate -- and really appreciate the help!

phil-opp, 2025-04-02:
It doesn't receive a stop message because the event stream is closed as soon as all inputs are closed. The stop message just signals a premature stop, e.g. triggered by ctrl-c. So if you change your condition, the loop should immediately exit (since there are no more inputs):

```diff
-         if event is not None and event["type"] == "STOP":
+         if event is not None:
            break
```

I guess some kind of `wait_for_manual_stop` input that never resolves would be useful. This could be used to keep the event stream open so that the `STOP` signal is received when ctrl-c is pressed.

pculbertson-bdai, 2025-04-08:
Yep, makes total sense. Will close now as a duplicate since this explains what's going on. Really appreciate the quick and detailed response, and thanks x1000 for developing this - I think it's going to be an amazing resource for the community. Y'all rock.

closed 2025-04-08 by pculbertson-bdai