dora-rs/dora #937: 只刷新一次的问题

G1017, 2025-04-02:
https://dora-rs.ai/docs/guides/getting-started/yolov8
根据这个修改了
plot.py
‘’‘
import os,time
import cv2
import gradio as gr

from dora import DoraStatus
from utils import LABELS
import pyarrow as pa

CI = os.environ.get("CI")

CAMERA_WIDTH = 640
CAMERA_HEIGHT = 480

FONT = cv2.FONT_HERSHEY_SIMPLEX

class Operator:
    """Plot image and bounding box."""

    def __init__(self):
        """TODO: Add docstring."""
        self.bboxs = []
        self.buffer = ""
        self.submitted = []
        self.lines = []

    def on_event(
        self,
        dora_event,
        send_output,
    ):
        """TODO: Add docstring."""
        if dora_event["type"] == "INPUT":
            id = dora_event["id"]
            value = dora_event["value"]
            image = None
            if id == "image":

                image = (
                    value.to_numpy().reshape((CAMERA_HEIGHT, CAMERA_WIDTH, 3)).copy()
                )

                for bbox in self.bboxs:
                    [
                        min_x,
                        min_y,
                        max_x,
                        max_y,
                        confidence,
                        label,
                    ] = bbox
                    cv2.rectangle(
                        image,
                        (int(min_x), int(min_y)),
                        (int(max_x), int(max_y)),
                        (0, 255, 0),
                    )
                    cv2.putText(
                        image,
                        f"{LABELS[int(label)]}, {confidence:0.2f}",
                        (int(max_x), int(max_y)),
                        FONT,
                        0.5,
                        (0, 255, 0),
                    )

                cv2.putText(
                    image, self.buffer, (20, 14 + 21 * 14), FONT, 0.5, (190, 250, 0), 1,
                )

                i = 0
                for text in self.submitted[::-1]:
                    color = (
                        (0, 255, 190)
                        if text["role"] == "user_message"
                        else (0, 190, 255)
                    )
                    cv2.putText(
                        image,
                        text["content"],
                        (
                            20,
                            14 + (19 - i) * 14,
                        ),
                        FONT,
                        0.5,
                        color,
                        1,
                    )
                    i += 1

                for line in self.lines:
                    cv2.line(
                        image,
                        (int(line[0]), int(line[1])),
                        (int(line[2]), int(line[3])),
                        (0, 0, 255),
                        2,
                    )

                if CI != "true":
                    pic_time = time.time()
                    pic_time_str = "pic/"+"image_" + str(pic_time) + ".jpg"
                    cv2.imwrite(pic_time_str, image)
                    image = image
                    # with gr.Blocks() as demo:
                    #     video_feed = gr.Image(label="视频流", interactive=False)  # 不能使用 streaming=True
                    #     demo.load(frame, inputs=None, outputs=video_feed)
                    # demo.launch(server_name="0.0.0.0", server_port=7860,share=True)
                    # cv2.imshow("frame", image)
                    # if cv2.waitKey(1) & 0xFF == ord("q"):
                    #     return DoraStatus.STOP
            elif id == "bbox":
                self.bboxs = value.to_numpy().reshape((-1, 6))
            elif id == "keyboard_buffer":
                self.buffer = value[0].as_py()
            elif id == "line":
                self.lines += [value.to_pylist()]
            elif "message" in id:
                self.submitted += [
                    {
                        "role": id,
                        "content": value[0].as_py(),
                    },
                ]
            if image is None:
                print("Error: Image is None!")
                return DoraStatus.CONTINUE
            send_output("frame",pa.array(image.ravel()),dora_event["metadata"],)

        return DoraStatus.CONTINUE

’‘’

mage_gr.py
‘’‘
import os,time
import cv2
import gradio as gr
import pyarrow as pa

from dora import DoraStatus
from utils import LABELS

CI = os.environ.get("CI")

CAMERA_WIDTH = 640
CAMERA_HEIGHT = 480

FONT = cv2.FONT_HERSHEY_SIMPLEX

class Operator:

    def __init__(self):
        self.buffer = ""
    def on_event(
        self,
        dora_event,
        send_output,
    ):
        """TODO: Add docstring."""
        if dora_event["type"] == "INPUT":
            id = dora_event["id"]
            value = dora_event["value"]
            if id == "frame":

                frame = (
                    value.to_numpy().reshape((CAMERA_HEIGHT, CAMERA_WIDTH, 3)).copy()
                )
                # frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
                with gr.Blocks() as demo:
                    video_feed = gr.Image(label="视频流", interactive=False)  # 不能使用 streaming=True
                    demo.load(lambda: frame, inputs=None, outputs=video_feed)
                demo.launch(server_name="0.0.0.0", server_port=7860,share=True)

        return DoraStatus.CONTINUE

’‘’

dataflow.yml
‘’‘
nodes:
  - id: webcam
    operator:
      python: webcam.py
      inputs:
        tick: dora/timer/millis/50
      outputs:
        - image

  - id: object_detection
    operator:
      send_stdout_as: stdout
      python: object_detection.py
      inputs:
        image: webcam/image
      outputs:
        - bbox
        - stdout

  - id: plot
    operator:
      python: plot.py
      inputs:
        image: webcam/image
        bbox: object_detection/bbox
        assistant_message: object_detection/stdout
      outputs:
        - frame

  - id: image_gr
    operator:
      python: image_gr.py
      inputs:
        frame: plot/frame
        
’‘’
但是gradio界面不刷新，只显示一张图片，请问如何修改

[image]

phil-opp, 2025-04-02:
Could you please translate this issue to English?

Jia-Baos, 2025-04-03:
@G1017  i think you can print the log in every node, so that you can know whether the data is properly and continuously transferred between nodes

ShashwatPatil, 2025-04-04:
Hey @G1017, I encountered a similar issue while trying to display a live video stream on gradio—only the first frame renders and then it freezes. From what I understand, Gradio doesn't currently support simply displaying a live video stream. I was able to use it to get a video stream input, but not just for display purposes. I'm still fairly new to Gradio, so I might be missing something—open to any suggestions!