dora-rs/dora #938: Debug Message Support

NageshMandal, 2025-04-02:
Allow any Dora node to send a text debug message and show it live in Rerun under a unique path like logs/<node_id>.
Add a new handler for inputs that contain "debug" or "log" in their ID.

NageshMandal, 2025-04-02:
@dora-bot assign me

ankitmrj, 2025-04-05:
Hi ,
This is my approach 
Steps to Implement
1. Locate the Input Handling Code
Look for the place in the Rerun codebase where it handles incoming inputs (maybe in a receiver, stream, or ingest module).
2.Add a Handler for Debug/Log Inputs
Modify the logic to check if the input id includes "debug" or "log"
3.. Create a Text Logging Utility (if needed)
If there isn't a built-in function to handle plain-text messages in Rerun yet, add one (maybe similar to how scalar/tensor/image logs are handled).

4. Ensure Live Update
Make sure that these logs are not just stored but pushed in real-time to the Rerun viewer. Use Rerun’s API for live updates or streaming.

5. Testing
Simulate a Dora node emitting a debug message with a known ID like debug_temperature_node.
Confirm that it appears in Rerun under logs/temperature_node.

I want to work on it.