dora-rs/dora #966: Benchmark & design issues

Mivik, 2025-04-15:
**Describe the bug**
As described in #760, benchmark ran on latest master doesn't seem good.

**To Reproduce**
`cargo run --example benchmark --release`

**Expected behavior**
A good bench.

**Environments (please complete the following information):**
 - System info: `Linux mivik-arch 6.14.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 10 Apr 2025 18:43:59 +0000 x86_64 GNU/Linux`
 - Dora version: latest master

**Additional context**

After investigating, I narrowed the issue down to this part:

https://github.com/dora-rs/dora/blob/b63b4c1b76ca57c0683aca132a7b1757b55b60ee/apis/rust/node/src/event_stream/mod.rs#L92-L100

So basically the default `queue_size` for any data id is `1`, causing the scheduler to discard any unhandled events on receiving new ones (**without any notice!**). This behavior caused the benchmark program to malfunction and eventually yield bad results.

After raising the `queue_size` for `throughput` the result seems better (but still weird though, needs further investigation):

```
Latency:
size 0x0     : 819.657µs
size 0x8     : 787.412µs
size 0x40    : 778.433µs
size 0x200   : 741.805µs
size 0x800   : 821.749µs
size 0x1000  : 796.586µs
size 0x4000  : 735.843µs
size 0xa000  : 767.344µs
size 0x64000 : 799.259µs
size 0x3e8000: 858.608µs
Throughput:
size 0x0     : 2080 messages per second
size 0x8     : 1490 messages per second
size 0x40    : 2289 messages per second
size 0x200   : 2308 messages per second
size 0x800   : 2441 messages per second
size 0x1000  : 2333 messages per second
size 0x4000  : 4483 messages per second
size 0xa000  : 238714 messages per second
size 0x64000 : 244618 messages per second
size 0x3e8000: 223972 messages per second
```

My questions are:

- Why such a small default queue size? Personally I deem it insufficient for the majority of intense cases.
- Shouldn't we give a warning or something on discarding events capped by queue size? Current behavior can be confusing.

If we reach a consensus I can assign myself on improving this behavior and fix benchmark issues.

haixuanTao, 2025-04-15:
Ok, thanks, good to know that this is related to the queue size. Would you mind opening a PR with your latest change?

> Why such a small default queue size? Personally I deem it insufficient for the majority of intense cases.

I think it depends if you want lowest latency or higest quality/resiliency. 

Adding a queue size usually means that if the processing time and receiving time is more or less equal you're adding the queue in latency on each input if you're process ever run late.

> Shouldn't we give a warning or something on discarding events capped by queue size? Current behavior can be confusing.

I think there.used to be warning. We can add it again.

Mivik, 2025-04-16:
> Would you mind opening a PR with your latest change?
> I think there.used to be warning. We can add it again.

See https://github.com/dora-rs/dora/pull/971

phil-opp, 2025-04-16:
> * Why such a small default queue size? Personally I deem it insufficient for the majority of intense cases.

Originally, the queue_size was set to 10 by default (see https://github.com/dora-rs/dora/pull/230 and https://github.com/dora-rs/dora/pull/231). It still seems to default to 10 at the daemon level, but we added some node-level queue cleanup in https://github.com/dora-rs/dora/commit/5ccdbc593ecd5a854e08f68254a3e3d684607dc9 that defaults to queue_size 1. This is the part that you linked above. I don't remember the motivation for using a different default queue size there, maybe this was just by accident...

> * Shouldn't we give a warning or something on discarding events capped by queue size? Current behavior can be confusing.

We discussed this a bit in https://github.com/dora-rs/dora/pull/533, but didn't reach a decision.

haixuanTao, 2025-04-17:
Basically anything other than 1, we'll eventually be reached and then you're running a queue_size * frequency latency, which can be very noticeable on any realtime use.

PR #971 (2025-04-22): Improve benchmark implementation & Add warning for discarding events
Resolves #966

This PR improves current benchmark implementation by increasing the queue size for throughput testing, and adding time gap between throughput tests of different sizes to ensure accurate measurement.

Besides, a warning message will be logged now on discarding events due to queue size limit.
commit 72b4be80 (2025-04-22): Improve benchmark implementation & Add warning for discarding events (#971)

Resolves #966

This PR improves current benchmark implementation by increasing the
queue size for throughput testing, and adding time gap between
throughput tests of different sizes to ensure accurate measurement.

Besides, a warning message will be logged now on discarding events due
to queue size limit.
closed 2025-04-22 by haixuanTao