dora-rs/dora #967: CI: `musllinux` job broken on `main` branch

phil-opp, 2025-04-15:
**Describe the bug**

All the `musllinux` jobs [fail on CI](https://github.com/dora-rs/dora/actions/runs/14432894203/job/40469831950#step:4:1704) on the `main` branch. The error message is:

```
 💥 maturin failed
  Caused by: Failed to run rustc to get the host target
  Caused by: error from command -- stderr:

sccache: error: failed to execute compile
sccache: caused by: Compiler not supported: "error: toolchain \'stable-x86_64-unknown-linux-gnu\' is not installed\nhelp: run `rustup toolchain install stable-x86_64-unknown-linux-gnu` to install it\n"
```

The first CI run that showed this error was https://github.com/dora-rs/dora/actions/runs/14432894203/job/40469831950 , but I don't think that it's related to the `crossbeam-channel` dependency update.

haixuanTao, 2025-04-16:
Seems to be a `sccache` issue. 

Easiest might be to disable it

phil-opp, 2025-04-16:
Thanks! I tried this in #901 and it seems to fix the error: https://github.com/dora-rs/dora/actions/runs/14497133129/job/40667600070?pr=901

haixuanTao, 2025-05-27:
This should be fixed as of #1000

closed 2025-05-27 by haixuanTao