dora-rs/dora #970: Way too many File Descripters when launch a keyboard node in dora v 0.3.11

jeffery4011, 2025-04-15:
**Describe the bug**
Hello, I am using a dora node to read my keyboard input with pynput.

But after starting my dora graph, the File Descripter (FD) I have becomes super large (~2000). This is blocking my access to Xlib:

2025-04-15T15:26:51.367198Z  INFO dora_daemon::log:       File "/home/qtus-szx/miniforge3/envs/policy/lib/python3.10/site-packages/Xlib/protocol/display.py", line 561, in send_and_recv
2025-04-15T15:26:51.367717Z  INFO dora_daemon::log: 01963a0f-6786-7581-8dc1-45e1a11a8470/keyboard info:
2025-04-15T15:26:51.367727Z  INFO dora_daemon::log:         rs, ws, es = select.select([self.socket], writeset, [], timeout)
2025-04-15T15:26:51.368254Z  INFO dora_daemon::log: 01963a0f-6786-7581-8dc1-45e1a11a8470/keyboard info:
2025-04-15T15:26:51.368264Z  INFO dora_daemon::log:     ValueError: filedescriptor out of range in select()

**To Reproduce**
Steps to reproduce the behavior:
I am running a graph with only one node:

`from pynput import keyboard
from pynput.keyboard import Key, Events
import pyarrow as pa
from dora import Node
import os

node = Node()
null_msg = pa.array([0])
print("[Keyboard] Keyboard node spinning...")
print("My FDs:", len(os.listdir(f'/proc/{os.getpid()}/fd')))
with keyboard.Events() as events:
    while True:
        event = events.get(0.1)
        # if event is not None and isinstance(event, Events.Press):
        #     if event.key == keyboard.KeyCode.from_char('l') or event.key == keyboard.KeyCode.from_char('L'):
        #         # print("L pressed")
        #         node.send_output("l_pressed", null_msg)
        # if event.key == Key.space and space == False:
        #     cursor += 1
        #     node.send_output("space", pa.array([cursor]))
        #     space = True
        if event is not None and isinstance(event, Events.Release):
            if event.key == Key.space:
                node.send_output("space", null_msg)
                print("space")
            elif event.key == Key.backspace:
                node.send_output("backspace", null_msg)
            elif event.key == Key.ctrl_l:
                print("ctrl_l")
                node.send_output("ctrl_l", null_msg)
            elif event.key == Key.esc:
                print("[Keyboard] Sending STOP signal to all nodes.")
                node.send_output("esc", null_msg)
                break
            elif event.key == Key.shift_l:
                print("shift_l")
                node.send_output("shift_l", null_msg)
            elif event.key == keyboard.KeyCode.from_char('l') or event.key == keyboard.KeyCode.from_char('L'):
                print("letter_l")
                node.send_output("letter_l", null_msg)
            # elif event.key == Key.enter:
            #     print("enter")
            #     node.send_output("enter", pa.array([0]))
        if node.next(0.001) is None:
            break
`

**Expected behavior**
In dora v 0.3.11, 
'print("My FDs:", len(os.listdir(f'/proc/{os.getpid()}/fd')))'
will output My FDs: 2328

and the dora node is not responding to my keyboard

but in dora v 0.3.8
the output is    My FDs: 8, the dora node is responding well

**Screenshots or Video**
If applicable, add screenshots to help explain your problem.

**Environments (please complete the following information):**
Ubuntu 22.04  64bit
Dora v 0.3.11
Python 3.10 

**Additional context**
Add any other context about the problem here.

alberthli, 2025-04-19:
I'm having a similar issue, where the issue arises when starting multiple ZED cameras. The number of files spikes to over 1000 after initializing a camera, and nearly immediately hits the internal limit of 1024 file descriptors. I don't know where the error originates from, but I instead get the following:
```
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
```

haixuanTao, 2025-04-20:
Ok interesting that it used to work at 0.3.8 and does not work since.

alberthli, 2025-04-20:
It’s worth noting that in my use case reverting to 0.3.8 doesn’t solve the issue - there’s possibly a bad interaction with the ZED SDK. However, if I just initialize a ZED camera object in the REPL, only on the order of a dozen FDs are used. In my dora stack, that explodes to 1-2k.

haixuanTao, 2025-05-27:
This issue should be fixed on main now thanks to @Mivik ! and the bump of the version of opentelemetry system metrics on #1008

closed 2025-05-27 by haixuanTao