dora-rs/dora #981: High-level system diagram and performance comparison to other run-time frameworks?

RobotLearner2022, 2025-04-24:
Really awesome work and hats to your outstanding team!

I have two (sets of) questions:
(1) Do you have any high-level system showing how each node sends and receives messages and keeps track of the timestamps? How are the messages in Arrow format buffered in the shared memory and eventually saved to on-disk parquet files?

(2) Have you done performance comparison to other run-time frameworks, such as Nvidia's Isaac ROS and Autoware Core? The specific metrics I'm interested in are the message timestamp reproducibility, overall frame rate, and compute resource requirements given the same input (say a driving log in a ROS2 bag or a scene from the open datasets, such as NuScenes or Waymo Open Datasets)? 

Thank you!

haixuanTao, 2025-04-26:
1. Is an awesome question and we should defintely work on something like this but we currently don't have anything of sort. We would love to build on top of opentelemetry tracing such system but I think there is still a bit of work to be done..

2. We haven't done the metric you have mentioned. We have been mostly interested in latency as it is currently for us the biggest low hanging fruit but we should definetely work on something for. 

I think 2 aim for more production system but the thing is we see that most of our users like to use python which by design is not made for those and so it's hard for us to focus on these at the moment.

Ultimately we hope that by making it easy for people to use rust we can have great system metrics as well.

RobotLearner2022, 2025-05-02:
Thanks for your response. Your system is indeed designed for users (including myself) to start prototyping in Python very quickly. Look forward to more exciting progress towards the industrial use cases!

closed 2025-05-02 by RobotLearner2022