dora-rs/dora #982: Problem: Frequent messages from sender node leading to data loss at receiver node.

chengzi0103, 2025-04-25:
**Describe the bug**

We have two nodes, 'a' and 'b'.

Node 'a' mimics the streaming output of the `python-openai-client`. It uses a generator function like the following to produce a stream of dictionaries with a specific structure (`type`, `content`, `articles`, `metadata`, `id`):

```python
def generate_stream():
    """
    A generator function that continuously returns dictionaries with a specific structure,
    including random content and incrementing IDs.
    """
    stage_id = 0  # Fixed stage ID
    sub_stage_counter = 0 # Sub-step counter for generating incrementing sub-stage IDs

    print(f"Structured generator started for stage {stage_id}...")

    while True:
        # Increment sub-step counter each loop
        sub_stage_counter += 1
        current_sub_stage_id = sub_stage_counter

        # 1. Generate random content for 'content' field
        # Example: String with random float and current timestamp
        random_chunk_content = f"Processing step {current_sub_stage_id}: random_value={random.random():.4f}, timestamp={time.time()}"
        # Other examples: random integer as string, randomly chosen word

        # 2. Populate 'articles' field
        # Simulates a list of dictionaries (a.dict())
        num_articles = random.randint(0, 4) # Generate 0 to 3 dummy articles randomly
        generated_articles = []
        for i in range(num_articles):
             generated_articles.append({
                 "title": f"Dummy Article {i+1} for step {current_sub_stage_id}",
                 "summary": f"Summary text with random number {random.randint(1, 100)}",
                 "source": random.choice(["Web", "Database", "LocalFile"])
             })

        # 3. Populate 'metadata' field
        stage_name = "Processing" # Example stage name

        # 4. Build 'id' field
        current_id = f"{stage_id}-{current_sub_stage_id}"

        # 5. Assemble the final dictionary structure
        output_data = {
            "type": "thinking",  # Fixed as "thinking" based on your example
            "content": random_chunk_content, # Generated random content
            "articles": generated_articles,    # Generated list of articles
            "metadata": {"stage": stage_name}, # Populated metadata
            'id': current_id                   # Constructed ID
        }

        # Yield the assembled dictionary
        yield output_data
```

Node 'a' then returns this stream of results to node 'b'.

Node 'b' is implemented using the code in this project location:
[https://github.com/moxin-org/mofa/tree/main/python/node-hub/openai-server-stream](https://github.com/moxin-org/mofa/tree/main/python/node-hub/openai-server-stream)

We observed during testing that when node 'a' returns the results too quickly (at a high frequency), node 'b' experiences significant data loss during reception. However, when we introduce a transmission interval of 0.005 seconds between messages from node 'a', this problem does not occur.

Our questions are:

* What is the underlying cause of this data loss issue?
* What are the recommended solutions to resolve this?

**Environments (please complete the following information):**
 - System info: mac os 
 - Dora version: dora-cli 0.3.10
 - Python Version: 3.12

eladb, 2025-04-25:
Have you tried to increase the `queue_size` on the consumer?

chengzi0103, 2025-04-25:
> Have you tried to increase the `queue_size` on the consumer?您是否尝试过增加消费者的 `queue_size` ？

We haven't tried using queue_size yet. We believe Dora should be able to handle this message rate. If we want to try configuring queue_size, how should we do that?

eladb, 2025-04-26:
You can adjust the queue_size for an input:

```yaml
  inputs:
    image:
      source: camera/image
      queue_size: 100
```

chengzi0103, 2025-04-27:
Ok, thanks, my problem is solved

closed 2025-04-27 by chengzi0103