dora-rs/dora #997: Enhancement Request: Unified and User-Friendly Metadata API in Rust

XiaoPengYouCode, 2025-05-07:
# Enhancement Request: Unified and User-Friendly Metadata API in Rust

## Issue Description

Currently, there's a significant discrepancy between the Python and Rust APIs for accessing metadata in dora-rs. The Python API provides a convenient dictionary-like interface for accessing metadata parameters with default values, while the Rust API requires manual handling of the `Parameter` enum variants and doesn't offer similar convenience methods.

### Current Python API Example
```python
# Convenient parameter access with default values
encoding = event["metadata"]["encoding"] # string
wait = event["metadata"].get("wait", True)  # bool
duration = float(event["metadata"].get("duration", 1))  # float or integer
...
```

### Current Rust API Usage
```rust
// Current approach requires manual type checking and unwrapping for each parameter
let encoding = if let Some(Parameter::String(val)) = metadata.parameters.get("encoding") {
    val.clone()
} else {
    "".to_string()
};

let wait = if let Some(Parameter::Bool(val)) = metadata.parameters.get("wait") {
    *val
} else {
    true
};

// And so on for each parameter...
```

## Proposed Enhancement

Add convenient accessor methods to the `Metadata` struct in Rust that provide a similar experience to the Python API. This would improve developer experience, reduce boilerplate code, and make the two APIs more consistent.

### Suggested Implementation

```rust
impl Metadata {
    // Generic getter that returns the raw Parameter
    pub fn get(&self, key: &str) -> Option<&Parameter> {
        self.parameters.get(key)
    }
    
    // Type-specific getters with default values
    pub fn get_string(&self, key: &str, default: &str) -> String {
        if let Some(Parameter::String(value)) = self.parameters.get(key) {
            value.clone()
        } else {
            default.to_string()
        }
    }
    
    pub fn get_bool(&self, key: &str, default: bool) -> bool {
        if let Some(Parameter::Bool(value)) = self.parameters.get(key) {
            *value
        } else {
            default
        }
    }
    
    pub fn get_int(&self, key: &str, default: i64) -> i64 {
        if let Some(Parameter::Integer(value)) = self.parameters.get(key) {
            *value
        } else {
            default
        }
    }
    
    pub fn get_float(&self, key: &str, default: f64) -> f64 {
        if let Some(Parameter::Float(value)) = self.parameters.get(key) {
            *value
        } else {
            default
        }
    }
    
    pub fn get_list_int(&self, key: &str, default: Vec<i64>) -> Vec<i64> {
        if let Some(Parameter::ListInt(value)) = self.parameters.get(key) {
            value.clone()
        } else {
            default
        }
    }
    
    pub fn get_list_float(&self, key: &str, default: Vec<f64>) -> Vec<f64> {
        if let Some(Parameter::ListFloat(value)) = self.parameters.get(key) {
            value.clone()
        } else {
            default
        }
    }
    
    pub fn get_list_string(&self, key: &str, default: Vec<String>) -> Vec<String> {
        if let Some(Parameter::ListString(value)) = self.parameters.get(key) {
            value.clone()
        } else {
            default
        }
    }
}
```

### Usage After Enhancement

```rust
// More convenient and readable parameter access
let encoding = metadata.get_string("encoding", "");
let wait = metadata.get_bool("wait", true);
let duration = metadata.get_float("duration", 1.0);
let grasp = metadata.get_bool("grasp", true);
let arm_p = metadata.get_string("arm", "right");
```

## Benefits

1. **Improved Developer Experience**: Reduces boilerplate code and makes the API more intuitive
2. **API Consistency**: Brings the Rust API closer to the Python API's convenience
3. **Safer Defaults**: Provides a standardized way to handle missing parameters
4. **Better Error Handling**: Encapsulates type conversion and error handling within the API
5. **Reduced Learning Curve**: Makes it easier for developers familiar with the Python API to use the Rust version

## Additional Considerations

- We could also consider adding conversion methods for more complex use cases
- A macros-based approach could potentially be used to make the implementation more concise
- Unit tests should be added to verify the behavior of these new methods

I believe this enhancement would significantly improve the usability of the Rust API while maintaining Rust's strong type safety guarantees.

haixuanTao, 2025-05-09:
Ok sure!

keirsalterego, 2026-03-09:
Hey, I went through the thread and PR #1337. Looks like TryFrom was added for type conversion but the default value part from the original issue is still missing right?
Was thinking of adding something like this on top of what's already there:
```
pub fn get_or<T>(&self, key: &str, default: T) -> T
where
    T: TryFrom<&'_ Parameter>,
{
    self.parameters
        .get(key)
        .and_then(|p| T::try_from(p).ok())
        .unwrap_or(default)
}
```
so you can just do metadata.get_or("wait", true) instead of the whole unwrap chain. Can I work on this @phil-opp?