dora-rs/dora #1001: Terminal logging freezes after ~30 seconds

yiyikkk, 2025-05-11:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. Start a new dataflow: `dora start dataflow.yaml`
3. Stop dataflow: `dora stop`
4. Destroy dataflow: `dora destroy`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or Video**
If applicable, add screenshots to help explain your problem.

**Environments (please complete the following information):**
 - System info: [use `uname --all` on Linux]
 - Dora version: [use `dora --version` and `pip show dora-rs`]

**Additional context**
Add any other context about the problem here.

yiyikkk, 2025-05-11:
**Additional Observation**
I also checked the system log file generated for this `keyboard` node:
Log path: `out/0196c19e-4e11-73dc-9d15-cfa96ffa7db2/log_keyboard.txt`

* Before I manually terminated the running dataflow with `Ctrl+C`, the `log_keyboard.txt` also froze after 30 log entries:

  ```
  Sending character: c
  Sending character: c
  Sending character: c
  ... (a total of exact 30 entries)
  ```
* After I pressed `Ctrl+C` to stop the dataflow, **all the pending logs were suddenly flushed and appended to the file**, like this:

  ```
  Sending character: c
  Sending character: c
  Sending character: c
  ... (a total of ~30 entries)
  [2m2025-05-11T23:12:10.962160Z[0m [33m WARN[0m [2mopentelemetry_system_metrics[0m[2m:[0m Could not get NVML, recording 0 for GPU memory usage
  Sending character: c
  Sending character: c
  Sending character: c
  ...
  (continued actual logs)
  ```

This suggests that not only the terminal output but also the internal log files are being blocked after a certain threshold, and they are only flushed upon shutdown.

haixuanTao, 2025-05-14:
Yes I agree!

Very annoying!

I wonder if this is linked to: Could not get NVML, recording 0 for GPU memory usage

sjfhsjfh, 2025-05-21:
Hi,

Just to add a bit more context: the 30s delay seems to come from the [`sleep` here](https://docs.rs/opentelemetry-system-metrics/0.3.1/src/opentelemetry_system_metrics/lib.rs.html#169).

Also, I noticed that using any `tracing` macro (like `warn!()`) in `Node` causes logs to be buffered and not flushed until the program exits. This might be because the `daemon` reads from `Node`'s stdout, and the default `tracing_subscriber::fmt()` uses a synchronous stdout writer, which can block in async contexts.

For what it's worth, I opened a PR upstream to replace the `sleep` with `interval.tick()` to avoid exposing such delays directly: https://github.com/haixuanTao/opentelemetry-system-metrics/pull/10

Let me know if there's more info on the intended logging design between `Node`, `Daemon`, and `Coordinator`—I'd be happy to help further or open a PR if needed.

sjfhsjfh, 2025-05-21:
As a workaround, simply replacing the `warn!()` call at [this line](https://github.com/haixuanTao/opentelemetry-system-metrics/blob/4db5246b477f92499b9fdbc20c83464e005c71ce/src/lib.rs#L252) with `tokio::io::stdout().write_all().await` can help avoid the issue.

**I believe this is not the correct way of doing this.**

haixuanTao, 2025-05-27:
This is temporarily fixed in #1008, with a long term improvement on #1009

commit 6619713a (2025-06-02): refactor(tracing): use builder style (#1009)

This PR refactors the tracing initialization to use the builder-style
API, improving configurability and clarity. Additionally, it removes the
blocking `std::io::stdout`-based tracing layer used in the `Node`, which
previously caused log output to hang in async contexts (Fix #1001).

Finally, several function names with misleading semantics have been
renamed for clarity.
closed 2025-05-27 by haixuanTao