dora-rs/dora #1003: Idea: Visual Graph Editor for YAML Configuration

royalrexwang, 2025-05-14:
**Is your feature request related to a problem? Please describe.**
I am a huge fan of Dora, and I have been studying some examples recently. I have some ideas regarding the configuration of Dora YAML files.
I'd like to propose an idea for a visual configuration editor for dora-rs, inspired by tools like [LabVIEW](https://en.wikipedia.org/wiki/LabVIEW) or [Blender Nods](https://docs.blender.org/manual/nb/2.79/render/blender_render/materials/nodes/introduction.html). Instead of writing dora.yaml manually, users could design the dataflow with drag-and-drop modules and export a valid YAML config.
**Describe the solution you'd like**
- Rust egui for visual configuration
-  Add/remove operator nodes
- Connect input/output ports between nodes
- Set node parameters via UI panel
- One-click export to dora.yaml
Thank you again for creating an open robotics framework. I'm excited to keep working with Dora, and hope this idea contributes toward its growth.

haixuanTao, 2025-05-14:
That would be awesome!

I agree that manually making dataflow is very annoying.

I think that this would work best as a standalone web-page of some sort.

There is a formal jsonschema specification of the yaml file here: https://github.com/dora-rs/dora/blob/cb53fc2c111829207f4a93907affee45894ccaad/libraries/core/dora-schema.json that you can have a look to be able to validate the yaml syntax

haixuanTao, 2025-05-21:
I wonder what a vibe coded visual graph editor would look like ! 

I believe it could be very very cool

heyong4725, 2025-05-21:
Take a look of this MoFA (an AI agent framework which is built on top of Dora) sub-project: https://github.com/moxin-org/mofa/tree/main/MoFA_stage, such that you can build your YAML declaration thru GUI.

phil-opp, 2025-11-26:
Very nice!