dora-rs/dora #1005: conversation demo pause after running for some seconds

stanal, 2025-05-15:
i'm trying the demo of https://dora-rs.ai/docs/guides/getting-started/conversation_py/ 
i can start it , but it only run for few seonds, and then has none ouput and log 

[image]

[image]

stanal, 2025-05-15:
[image]
the dataflow is still running when check dora list

stanal, 2025-05-15:
[image]

ShashwatPatil, 2025-05-20:
this should not happen, could you share you dataflow.yml if not same as in the guide, since i was not able to reproduce this error

closed 2025-05-27 by haixuanTao