dora-rs/dora #1110: opentelemetry version conflict

lijingrs, 2025-08-27:
cargo build
   Compiling dora-metrics v0.3.12
error[E0308]: mismatched types
   --> /Users/lijing/.cargo/registry/src/rsproxy.cn-e3de039b2554c837/dora-metrics-0.3.12/src/lib.rs:42:27
    |
42  |     init_process_observer(meter).await.unwrap();
    |     --------------------- ^^^^^ expected `opentelemetry::metrics::meter::Meter`, found `Meter`
    |     |
    |     arguments to this function are incorrect
    |
note: two different versions of crate `opentelemetry` are being used; two types coming from two different versions of the same crate are different types even if they look the same
    |
   ::: /Users/lijing/.cargo/registry/src/rsproxy.cn-e3de039b2554c837/opentelemetry-0.30.0/src/metrics/meter.rs:296:1
    |
296 | pub struct Meter {
    | ^^^^^^^^^^^^^^^^ this is the expected type `opentelemetry::metrics::meter::Meter`
   --> /Users/lijing/.cargo/registry/src/rsproxy.cn-e3de039b2554c837/opentelemetry-0.29.1/src/metrics/meter.rs:296:1
    |
296 | pub struct Meter {
    | ^^^^^^^^^^^^^^^^ this is the found type `Meter`
    |
   ::: /Users/lijing/.cargo/registry/src/rsproxy.cn-e3de039b2554c837/dora-metrics-0.3.12/src/lib.rs:14:5
    |
14  | use opentelemetry::{global, InstrumentationScope};
    |     ------------- one version of crate `opentelemetry` used here, as a direct dependency of the current crate
...
17  | use opentelemetry_system_metrics::init_process_observer;
    |     ---------------------------- one version of crate `opentelemetry` used here, as a dependency of crate `opentelemetry_system_metrics`
    = help: you can use `cargo tree` to explore your dependency tree
note: function defined here
   --> /Users/lijing/.cargo/registry/src/rsproxy.cn-e3de039b2554c837/opentelemetry-system-metrics-0.4.3/src/lib.rs:75:14
    |
75  | pub async fn init_process_observer(meter: Meter) -> Result<()> {
    |              ^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0308`.
error: could not compile `dora-metrics` (lib) due to 1 previous erro

information about this error, try `rustc --explain E0308`.
error: could not compile `dora-metrics` (lib) due to 1 previous error
> cargo tree
├── dora-node-api v0.3.12
│   ├── aligned-vec v0.5.0
│   │   └── serde v1.0.219
│   │       └── serde_derive v1.0.219 (proc-macro)
│   │           ├── proc-macro2 v1.0.101
│   │           │   └── unicode-ident v1.0.18
│   │           ├── quote v1.0.40
│   │           │   └── proc-macro2 v1.0.101 (*)
│   │           └── syn v2.0.106
│   │               ├── proc-macro2 v1.0.101 (*)
│   │               ├── quote v1.0.40 (*)
│   │               └── unicode-ident v1.0.18
│   ├── arrow v54.3.1
│   │   ├── arrow-arith v54.3.1
│   │   │   ├── arrow-array v54.3.1
│   │   │   │   ├── ahash v0.8.12
│   │   │   │   │   ├── cfg-if v1.0.3
│   │   │   │   │   ├── getrandom v0.3.3
│   │   │   │   │   │   ├── cfg-if v1.0.3
│   │   │   │   │   │   └── libc v0.2.175
│   │   │   │   │   ├── once_cell v1.21.3
│   │   │   │   │   └── zerocopy v0.8.26
│   │   │   │   │   [build-dependencies]
│   │   │   │   │   └── version_check v0.9.5
│   │   │   │   ├── arrow-buffer v54.3.1
│   │   │   │   │   ├── bytes v1.10.1
│   │   │   │   │   ├── half v2.6.0
│   │   │   │   │   │   ├── cfg-if v1.0.3
│   │   │   │   │   │   └── num-traits v0.2.19
[... 716 lines cut ...]
│   │   ├── eyre v0.6.12 (*)
│   │   ├── raw_sync_2 v0.1.5
│   │   │   ├── cfg-if v1.0.3
│   │   │   ├── libc v0.2.175
│   │   │   └── nix v0.23.2
│   │   │       ├── bitflags v1.3.2
│   │   │       ├── cfg-if v1.0.3
│   │   │       ├── libc v0.2.175
│   │   │       └── memoffset v0.6.5
│   │   │           [build-dependencies]
│   │   │           └── autocfg v1.5.0
│   │   ├── serde v1.0.219 (*)
│   │   ├── shared_memory_extended v0.13.0
│   │   │   ├── cfg-if v1.0.3
│   │   │   ├── libc v0.2.175
│   │   │   ├── nix v0.26.4
│   │   │   │   ├── bitflags v1.3.2
│   │   │   │   ├── cfg-if v1.0.3
│   │   │   │   └── libc v0.2.175
│   │   │   └── rand v0.8.5 (*)
│   │   └── tracing v0.1.41 (*)
│   ├── shared_memory_extended v0.13.0 (*)
│   ├── tokio v1.47.1 (*)
│   └── tracing v0.1.41 (*)
└── rand v0.8.5 (*)

phil-opp, 2025-08-27:
Thanks for reporting! I just opened PR  https://github.com/dora-rs/dora/pull/1109 to update our dependencies and fix some version mismatches, including some opentelemetry ones. Could you try that branch and see whether this fixes your issue? Thanks!

haixuanTao, 2025-08-27:
Sorry it's probably because of me!

haixuanTao, 2025-08-30:
FYI I have yanked the open telemetry-system-metrics to fix this

commit b2eef142 (2025-09-16): Downgrade `opentelemetry` to `v0.29` again in `dora-metrics` (#1132)

Version `v0.4.3` of `opentelemetry-system-metrics`, which updated to
`opentelemetry v0.30`, was yanked (as it violated semver). This causes
some build errors when dataflows/nodes are created through `dora new`
because those then use `opentelemetry-system-metrics v0.4.2` again,
which requires `opentelemetry v0.29`.

This PR fixes this issue by downgrading the `opentelemetry` dependencies
to v0.29 again (until `opentelemetry-system-metrics v0.5` is published).

Fixes #1110
closed 2025-09-03 by phil-opp